Atmosphere-NX/Atmosphere #872: i cannot play tales of vesperia on emuMMC

novalkun, 2020-04-11:
## Bug Report

### What's the issue you encountered?

i copied my sysMMC to emuMMC with partition based. i did not make any changes to atmosphere.
on sysMMC i can play Tales of Vesperia well, but **in emuMMC, the game can be loaded but it will stuck on language selection screen**.

### How can the issue be reproduced?
-
### Crash Report
-
### System Firmware Version

9.2.0 (exfat, but my SD now Fat32)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

Kosmos

### Additional context?

ELY3M, 2020-04-11:
can you try stock atmosphere?     rename your atmosphere folder to atmosphere_old 
download the zip from https://github.com/Atmosphere-NX/Atmosphere/releases  
unzip and copy atmosphere folder to your sd card.  then inject with latest fuse-primary.bin to boot atmosphere.

novalkun, 2020-04-11:
> can you try stock atmosphere? rename your atmosphere folder to atmosphere_old
> download the zip from https://github.com/Atmosphere-NX/Atmosphere/releases
> unzip and copy atmosphere folder to your sd card. then inject with latest fuse-primary.bin to boot atmosphere.

i tried your suggestion with atmosphere. it's the same, sadly

Buzbee, 2020-04-12:
I had the same problem on sysnand with 90dns enabled. The workaround was to turn off wifi completely and start the game.

novalkun, 2020-04-14:
> I had the same problem on sysnand with 90dns enabled. The workaround was to turn off wifi completely and start the game.

i have turned airplane mode on to the switch and the game runs perfectly. weird..... this wifi would be the criminal .......

closed 2020-04-20 by SciresM