Atmosphere-NX/Atmosphere #873: Black screen

Stue1975, 2020-04-11:
## Bug Report

N/A to all 

### What's the issue you encountered?

When switch is entered into dock the imagine dose not transfer onto my year old TV just get a no signal black image, tried on two different TV's and HDMI cables in Atmosphere, 

If started "normally" picture appears without an issue.

### How can the issue be reproduced?

Different hardware replicates the issue while in Atmosphere.

### Crash Report

### System Firmware Version

9.0.1

### Environment?

- What bootloader hekate
-
- Homebrew software installed: 

KOSMOS

### Additional context?

Please help me get the SW connected via dock to my LG TV,  Everything works perfectly apart from getting it on the TV.  I've tried a LG and a Sony TV and both while booting to Atmosphere goes Black on the TV........ handheld is always perfect.

Booting "normally" Both TV and handheld is perfect.

Any help would be so much appreciated. 

Kind regards

deadair3210, 2020-04-11:
Sounds like a issue with your dock or your hdmi port/cords
Also always try issues with the official build of Atmosphere to make sure the issue persists

Stue1975, 2020-04-11:
Many thanks for taking the time to reply.. iam confused how it may be the dock or hdmi port/cards as while in NSW FW all is fine.

Only in CFW the issue starts...  

Many thanks

deadair3210, 2020-04-11:
It might not be your dock, might be the switch itself. All I know is my Switch is running the latest official release and works fine in my dock

Stue1975, 2020-04-11:
Hi 

Yes that may be the case ill keep investigating.... thinking what else i could try.

thanks

Stue1975, 2020-04-12:
Further investigation:

While in CFW cartridge games work fine, must be a software issue with the "other" games and how they run via the CFW software

CrypticHelix, 2020-04-17:
Just wanted to say that I am having the same problem. I've tried reformatting the card and loading the software back onto it from the backup folder on my PC but nothing seems to get it past the black screen. Running 10.0.0

veesonic, 2020-04-22:
This behavior sounds similar to something I am experiencing where I cannot boot into emummc. I use Kosmos and added an issue to their GitHub.

https://github.com/AtlasNX/Kosmos/issues/274

closed 2020-05-07 by SciresM