Atmosphere-NX/Atmosphere #874: [dmntcht] Expose svcContinueDebugEvent and svcBreakProcess functions

olliz0r, 2020-04-11:
## Feature Request
### What feature are you suggesting?
#### Overview:
- Extend dmntcht to allow other sysmodules/homebrew to call svcBreakProcess and svcContinueDebugEvent on the debugged process</br>

#### Smaller Details: 
- N/A</br>

#### Nature of Request:  
- Addition
  - Extending the dmntcht service by two functions, exposing the svcBreakProcess and svcContinueDebugEvent functions on the debughandle.

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - dmnt</br>

### Why would this feature be useful?
This would allow homebrew applications and other sysmodules to freeze and unfreeze the process before writing to/reading from it to guarantee, that the application does not modify the data before the operation has completed. 

Example: Using dmntcht right now it is not possible to inject a full box of pokemon in pokemon sword/shield using a sysmodule while the game is running, without the game noticing half written slots and "fixing" them, resulting in broken data (bad eggs). Going the "original" route of calling svcDebugActiveProcess, svcWriteDebugProcessMemory and CloseHandle works fine, however, because the process stays frozen during the writing.</br>

SciresM, 2020-04-17:
https://github.com/Atmosphere-NX/Atmosphere/pull/899

SciresM, 2020-04-17:
Please verify that this works/meets your needs, @olliz0r.

SciresM, 2020-04-17:
Also pinging @AnalogMan151 in case he has any interest in testing.

SciresM, 2020-04-25:
Verified to work by analogman.

closed 2020-04-25 by SciresM