Atmosphere-NX/Atmosphere #876: Crash when scrolling past parental controls

Lapan, 2020-04-12:
## Bug Report
</br>

### What's the issue you encountered?

[ * ] Crash when scrolling past parental controls in the settings if Incognito RCM is active and parental controls are enabled.</br>

### How can the issue be reproduced?

[ * ]activate parental controls</br>
[ * ]use incognito rcm</br>
### Crash Report
https://gist.github.com/Lapan/6748c422d6d9f420b4a975b0c4bacbdb</br>

### System Firmware Version

9.2.0</br>

### Environment?

Kosmos

### Additional context?

N/A

SciresM, 2020-04-12:
That crash is in the home menu, error code is an ssl error code.

This is not an atmosphere problem, and incognito is not supported by Atmosphere.

closed 2020-04-12 by SciresM

SciresM, 2020-04-12:
To be more clear about this:

That's pctl/home menu failing because the ssl cert is invalid.

Incognito's whole thing is it makes the ssl cert invalid.

There's no issue to resolve here, because the user caused it themselves...the options are don't use incognito or don't use pctl I guess.