Atmosphere-NX/Atmosphere #877: The version has not changed

orelumbras, 2020-04-14:
I transferred the Atmosphere Files 0.10.5 files to my SD card with the hack but after that nothing happened. I restored the backup I did because nothing happened. The version wasn't updated. What can I do? (I have atmosphere 0.8.3)
Maybe I didn't do the process well either? Can you send how to update? 
Thx :)

nostone123, 2020-04-14:
did you happen to update your system firmware to version 10? if you did then you need to wait for a new update for atmosphere, this was for version 9.2

orelumbras, 2020-04-14:
My previous program is 8.1.0 and the atmosphere is 0.9.3.  Is there an update to my atmosphere version or should I wait?

orelumbras, 2020-04-14:
@nostone123 
My program firmware is 8.1.0.  The atmosphere is 0.9.3.  Is there an update to my atmosphere version or should I wait?

nostone123, 2020-04-14:
Oh should have worked then(the latest release thats currently out), are you using the current fusse-primary.bin file to inject?

orelumbras, 2020-04-14:
Oh, need to use it?  Can you list me step by step how to update? Because I'm new with the atmosphere and no one explained to me how to update...

nostone123, 2020-04-14:
What method do you use to inject your payloads?

orelumbras, 2020-04-14:
How can I check it?

orelumbras, 2020-04-14:
I know I have a pegascape caffeine, and I'm not allowed to be online

orelumbras, 2020-04-14:
And atlasNX KOSMOS

SciresM, 2020-04-14:
Reopen with issue templates and full sentences.

closed 2020-04-14 by SciresM