Atmosphere-NX/Atmosphere #881: Software Closes while on new 0.11.0

ReZeroEmiria, 2020-04-14:
Accidentally updated my Switch to 10.0 yesterday, updated Atmosphere to 0.11.0 now to try it out, it boots fine however when I play Animal Crossing shortly after getting into the game it will close the game out and say the software has stopped due to an error. No error code or anything.

Trying to play animal crossing now without Atmosphere running and it seems to work, I've been here much longer without getting booted from the game.

SciresM, 2020-04-14:
Could you try re-downloading the release build? I've seen some other reports like this but the errors don't make very much sense.

I disabled kernel loader as a preventative measure "just in case" it's causing problems in some way I don't understand, so I'm interested in whether the new build still exhibits the problem.

The build with the "just in case" fix is `-0775c788`.

stallionoid, 2020-04-14:
That seemed to fix it for me, idk about OP.

SciresM, 2020-04-14:
Sadly, it seems super random (per-boot-of-the-game?).

stallionoid, 2020-04-14:
At least if it's inconsistent rebooting can be a temporary bandaid fix? At least until a real fix is found

ReZeroEmiria, 2020-04-14:
Going to test playing some with the new build, before it booted me out every time at different intervals, though it became really severe after like 2 tries (1 min of playtime and boot)

SciresM, 2020-04-14:
This is an exosphere issue. I am unclear how it is, but it cannot be anything else -- I reproduced a crash booting completely stock other than exosphere.

johnnycanal, 2020-04-14:
I too have the crash on games like animal crossing new horizons sometimes on boot sometimes when connection online

ReZeroEmiria, 2020-04-14:
I've heard of exosphere being in the works from your streams and such but am not very familiar with what it is exactly. Is it necessary to use this exosphere that is causing the problem or can you release a build without it temporarily while the issue is worked on?

SciresM, 2020-04-14:
Heh. Unfortunately, exosphere is the one component that I literally cannot swap out, no.

ReZeroEmiria, 2020-04-14:
I see, so then there's some sort of interaction between it and the new 10.0 firmware causing this? I'm assuming the exosphere component was also in the previous build of Atmosphere.

SciresM, 2020-04-14:
No, the issue also occurs on < 10.0.0.

ReZeroEmiria, 2020-04-15:
So then it's just an issue with the new build?  hmm

lludu, 2020-04-15:
also having this issue, seems more common when trying to connect online into a game.

ThatAnarchyDude, 2020-04-15:
I can't even get it to boot the CFW.
I get the usual crash screen right after the Switch logo.
[image]

Lorenzooone, 2020-04-15:
Don't know if this can help, however with the old update I had issues with flickering textures in AC:NH. (Before it crashed)

Lorenzooone, 2020-04-15:
Update: the crashes still happen with the new download.

johnnycanal, 2020-04-15:
Just wait for an update and be patient thats why its still in pre release obviously its not to 100% the dev already took notice and are working on it

Lorenzooone, 2020-04-15:
It's not about being impatient. The more info a dev gets, the better. If the crashes still happen with the new build, it's better that they know.

ReZeroEmiria, 2020-04-15:
> Don't know if this can help, however with the old 0.11.0 update I had issues with flickering textures in AC:NH. (Before it crashed)

What were you doing when the flickering textures were occuring? Where were you in game?

SciresM, 2020-04-15:
fwiw, a lot of diagnosing has been going on on discord instead of in the issue tracker. Flickering textures are easily seen with blathers in AC.

ThatAnarchyDude, 2020-04-15:
Where is the Discord, please?
I thought I saw a link around somewhere while I was researching issues with the new updates, but I can't find it now.

SciresM, 2020-04-15:
there is a link in the issue templates.

oceanmaster41, 2020-04-15:
Best way to use as of now to downgrade 0.9.2. And using of 0.10.5 cfw. Very easy to do.

SuperN0va, 2020-04-15:
I‘m using the latest release, and the ACNH crushed after loading, then jumps out the error said the game data broken’

closed 2020-04-15 by SciresM