Atmosphere-NX/Atmosphere #882: Atmosphere crashing upon attempting to load the home menu.

Nova2127u, 2020-04-14:
So i updated everything to the latest version (Atmosphere, my Switch, and the payload) but after it loading into the home menu the Switch just crashes, i can load my Switch just fine without Atmosphere but when i try to use Atmosphere to load through it just crashes. i had custom themes but i have removed them from the sd card when this happened. Any help would be apperciated.
[image]

SciresM, 2020-04-14:
This is caused by your custom theme. Please remove it.

closed 2020-04-14 by SciresM

Nova2127u, 2020-04-14:
How can i remove it? i do not see anywhere on how to remove it.

SciresM, 2020-04-14:
Delete the /atmosphere/contents/0100000000001000 folder.

Nova2127u, 2020-04-14:
Alright, Thank you.