Atmosphere-NX/Atmosphere #883: Black screen

3dgray, 2020-04-14:
Accidentally updated to 10.0 yesterday, I've moved Atmosphere 0.11.0 and fusee-primary to SD. After injecting the payload I get a black screen after the two atmosphere startup screens

Philipp-94, 2020-04-14:
Are you using CFW-sys? If so, I second this, keeps booting until the atmosphere screen, black afterwards.

If you're using emunand, maybe a theme does crash the system. Go try to delete the "0100000000001000" Folder inside sdcard/atmosphere/contents/ .
That worked for my second (emunand) sd card.

Sysnand though, I hope for a solution too.

Killakelv, 2020-04-14:
I also accidently updated to software 10 & downloaded this and atomsphere boot screen showed and then black screen

Commander-Crash, 2020-04-15:
Atmosphere has no support for Switch FW10 I did the same and it will tell me "unknown pkg1"

ThatAnarchyDude, 2020-04-15:
I tried to update Atmosphere & Hekate.
I get to the Switch logo, and am greeted by this.
[image]

Lorenzooone, 2020-04-15:
Try wiping the atmosphere folder. It worked for me.

ThatAnarchyDude, 2020-04-15:
That's where all of my gamesaves are...28Gb.
That will take me hours to backup & transfer back.

shchmue, 2020-04-15:
> That's where all of my gamesaves are...28Gb.
> That will take me hours to backup & transfer back.

game saves shouldn’t be in the atmosphere folder. do what the user above did and delete your theme folder in atmosphere/contents/0100000000001000

Philipp-94, 2020-04-15:
Deleting everything from atmosphere folder and using ONLY the atmosphere from the release solved the problem for me. Don't know why, smth must've cross shooted between files I had inside previously. All is there, games, saves, etc. Now it's sigpatches which are missing. 

So at least to me, problem is solved. I can boot cfw-sys again.

SciresM, 2020-04-15:
issue isn't solved, it's just random and shitty.

3dgray, 2020-04-15:
I took the advice of deleting the entire atmosphere folder and using the new build. I can now boot into cfw. thank you

closed 2020-04-15 by 3dgray

alejandroalmarza, 2020-05-31:
Hello. I got a similar problem but the issue occurs when i install a nsp on the sd card. it work fine as long that i install on the emummc inner memory. there is my error log.
[Atmosphere fatal error 0x177A02 (2002-3005).txt](https://github.com/Atmosphere-NX/Atmosphere/files/4707808/Atmosphere.fatal.error.0x177A02.2002-3005.txt)

each time i have this error the only thing i can do to solve its creating a new emummc. Please help me