Atmosphere-NX/Atmosphere #884: Black Screen Upon Boot Into 0.11.0

poiagoboy, 2020-04-14:
## Bug Report
I accidentally update yesterday and when I send the new fuseprimary over, my screen displays black while doing nothing. The display is on as I can see the backlighting but will not load into Atmosphere. I had no issues when first getting into RCM mode again.

poiagoboy, 2020-04-14:
> ## Bug Report
> I accidentally update yesterday and when I send the new fuseprimary over, my screen displays black while doing nothing. The display is on as I can see the backlighting but will not load into Atmosphere. I had no issues when first getting into RCM mode again.

please disregard, my problem was solved

closed 2020-04-14 by poiagoboy

3dgray, 2020-04-14:
How did you solve this please?

poiagoboy, 2020-04-14:
I did not previously install all aspects of the new release. Once I did
that my problem was solved. However now I cannot open any applications

3dgray, 2020-04-14:
May I ask what you mean by installing all aspects?
I have placed the new update into the root of my SD card and overwritten the previous files of the same type and placed the fusee-primary in the payloads folder. ALso I have updated Hekate, what have I missed?

poiagoboy, 2020-04-15:
I completely removed atmosphere from my SD card and installed the new
build. I can now boot into atmosphere properly but cannot load any of my
apps or games nor can I hold Rb when starting an app to enter the homebrew
menu with full ram. I can still access homebrew through the Lb album entry

3dgray, 2020-04-16:
Unfortunately that will be because the keys haven’t been updated yet. I just downgraded my switch back to 9.2.0 in the end, it only took 10 mins and I didn’t burn a fuse

poiagoboy, 2020-04-17:
Is there an expected time frame on the new keys?

3dgray, 2020-04-17:
I couldn’t say, I didn’t want to wait

shchmue, 2020-04-17:
there are no new keys.