Atmosphere-NX/Atmosphere #887: Not Booting Past the Switch Logo

ThatAnarchyDude, 2020-04-15:
## Bug Report

### What's the issue you encountered?

I replaced all existing Atmosphere files with the new ones @SciresM posted on Twitter today.
I replaced all of the existing Hekate files with the new ones @CTCaer posted on Twitter today.

### How can the issue be reproduced?

I attempted to boot the standard CFW (firmware boot in between the EmuNAND & Stock choices on the menu), and I got all the way to the Switch logo before being presented with this.
[image]

### Crash Report

Crash Reports:
[01586912723_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478692/01586912723_0100000000000014.log)
[01586912724_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478693/01586912724_0100000000000014.log)
[01586912860_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478694/01586912860_0100000000000014.log)

### System Firmware Version

10.0.0</br>

### Environment?

Atmosphere was downloaded from
[The @SciresM Twitter Post today, April 14, 2020](https://twitter.com/SciresM/status/1250176567021457411?ref_src=twsrc%5Etfw)
and was launched by latest Hekate (hekate_ctcaer_5.1.4.bin) downloaded from
[The @CTCaer Twitter Post today, April 14, 2020](https://twitter.com/SciresM/status/1250176567021457411?ref_src=twsrc%5Etfw)

### Additional context?

Would my Fatal Reports logs help?
Fatal Reports:
[01584215621_0100000000001007.log](https://github.com/CTCaer/hekate/files/4478697/01584215621_0100000000001007.log)
[01586911481_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478698/01586911481_0100000000001000.log)
[01586911509_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478699/01586911509_0100000000001000.log)
[01586912548_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478700/01586912548_0100000000001000.log)
[01586912708_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478701/01586912708_0100000000001000.log)
[01586912782_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478702/01586912782_0100000000001000.log)
[01586913846_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478703/01586913846_0100000000001000.log)
[01586914015_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478704/01586914015_0100000000001000.log)
[01586914164_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478705/01586914164_0100000000001000.log)
[01586914617_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478706/01586914617_0100000000001000.log)
[01586914657_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478707/01586914657_0100000000001000.log)
[01586914690_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478708/01586914690_0100000000001000.log)
[01586914727_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478709/01586914727_0100000000001000.log)
[01586914754_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478710/01586914754_0100000000001000.log)

I will supply the Crash/Fatal dumps as well, if that would help.
I've never had to post an issue with Atmosphere or Hekate before...so I'm not really sure where to go from here or what else I can do to help.
Thank you.

athicom, 2020-04-15:
see "Black screen #883"

ThatAnarchyDude, 2020-04-15:
Wiping the Atmosphere folder?
That's where all of my gamesaves are...28Gb.
That will take me hours to backup & transfer back.

Commander-Crash, 2020-04-15:
I just updated my bootloader folder with latest from hekate. It allows me to boot into fusee-primary but still can't boot into emmuCFW

ThatAnarchyDude, 2020-04-15:
I can't boot into anything...except back into Hekate, I think.

ThatAnarchyDude, 2020-04-15:
There's the menu with 3 choices:
EmuMMC
Atmosphere (CFW)
Stock Firmware

I can't get ANY of them to boot from that menu.

Jboy93, 2020-04-15:
I have a similar issue. Updated hekate to 5.1.4 and atmosphere to 0.11.1 with my switch on 10.0.0 but can't seem to get past the atmosphere boot logo after choosing the fusee primary bin file. All I get is a black screen and nothing is happening.....

Jboy93, 2020-04-15:
Now that I finally boot into the cfw thanks to #883  I can't seem to play any games....it says the software cannot start....

masagrator, 2020-04-15:
Piracy is not supported

MegAmi24, 2020-04-15:
> Piracy is not supported
Where did anyone say they were pirating games

veesonic, 2020-04-17:
I have also been experiencing similar behavior. I use Kosmos and reported an issue on their github. I have a screenshot of what the log shows when attempting to boot into emummc, I also get a black screen.

https://github.com/AtlasNX/Kosmos/issues/274

(It looks like there were a couple of duplicate issues. Sorry if posting this is against rules, this comment can be deleted if needed.)

SciresM, 2020-04-18:
>qlaunch crash

Delete your custom theme/whatever else you have in the contents/0100000000001000 folder.

closed 2020-04-18 by SciresM

alejandroalmarza, 2020-05-31:
> > qlaunch crash
> 
> Delete your custom theme/whatever else you have in the contents/0100000000001000 folder.

Hello. i tried eliminating the atmosphere folder and using the original folder but it didn´t work. 
I can use the emummc with no problems until I install a game on the sd. then when I try to initiate the game it show an atmosphere error and then the emummc get stuck on the switch logo. The only solution is to create a new emummc and install games only in the emummc partition but it its too small only 22Gb of free space.

I am using fw 10.0.3 and atmosphere 0.12

this is my error log
`Atmosphère Fatal Report (v1.1):
Result:                          0x177A02 (2002-3005)

Program ID:                      0100000000001007
Process Name:                    playerSelec
Firmware:                        10.0.3 (Atmosphère 0.12.0-master-7bc0250c)
General Purpose Registers:
Start Address:                   0000005cd0600000
Stack Trace:
        ReturnAddress[00]:       0000005cd07435d8
        ReturnAddress[01]:       0000005cd075f5b0
        ReturnAddress[02]:       0000005cd075f710
        ReturnAddress[03]:       0000005cd075f6a8
        ReturnAddress[04]:       0000005cd075f328
        ReturnAddress[05]:       0000005cd09ee0d8
        ReturnAddress[06]:       0000005cd09c84c0
        ReturnAddress[07]:       0000005cd09cca8c
        ReturnAddress[08]:       0000005cd09dfec0
        ReturnAddress[09]:       0000005cd09f2a50
        ReturnAddress[10]:       0000005cd0844e98
        ReturnAddress[11]:       0000005cd0861b40
        ReturnAddress[12]:       0000005cd0861258
        ReturnAddress[13]:       0000005cd07373a8
        ReturnAddress[14]:       0000005cd07398c4
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000
        ReturnAddress[21]:       0000000000000000
        ReturnAddress[22]:       0000000000000000
        ReturnAddress[23]:       0000000000000000
        ReturnAddress[24]:       0000000000000000
        ReturnAddress[25]:       0000000000000000
        ReturnAddress[26]:       0000000000000000
        ReturnAddress[27]:       0000000000000000
        ReturnAddress[28]:       0000000000000000
        ReturnAddress[29]:       0000000000000000
        ReturnAddress[30]:       0000000000000000
        ReturnAddress[31]:       0000000000000000

[Atmosphere fatal error 0x177A02 (2002-3005).txt](https://github.com/Atmosphere-NX/Atmosphere/files/4707832/Atmosphere.fatal.error.0x177A02.2002-3005.txt)

Please I need help