Atmosphere-NX/Atmosphere #888: Game crashed on certain software

redondo56, 2020-04-15:
## Bug Report

Whenever I start up The Legend of Zelda Breath of the wild with version 0.11.0, the game starts running, starts flickering, and then after a minute the game crashes and it happens every time I boot the game up with the new version and it didn't happen before though with 0.10.5. I was wandering what the problem was and decided to reported after trying a few things. Every other game works fine though, even with games in which I have mods on but only Breath of the wild has been crashing after a few seconds. Was wondering if there was a way to fix this, thank you and I'm very sorry if I wasted your time.

AnalogMan151, 2020-04-15:
This is the same as issue #881, it's happening with multiple games. Flickering of some part of the game seems to be a common issue before the crash.

redondo56, 2020-04-15:
Yeah, the game runs normally at first then ends the screen starts getting corrupted very quickly until it crashes in very little amount of time. Thanks you for telling me about it so quickly by the way.

closed 2020-04-15 by SciresM