Atmosphere-NX/Atmosphere #890: An error after latest Atmosphere update.

Arathriel, 2020-04-15:
At the beginning I want to mention I'm using RCMloader and jig, maybe it's important. 

It's the first time for me seeing this bug and I'm using Atmopshere for 3-4 months already. I did everything as always, copied .zip folder and fusee-primary, then pasted to my sd-card. After Atmopshere logo etc i have a black screeen for couple secs and an error:

A fatal error occurred running Atmosphere.
Title ID:    010000000000002b
Error Desc: std : : abort ( )   called   (0xffe)
Report saved to /atmosphere/fatal_errors/

[Error.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4480220/Error.zip)

masagrator, 2020-04-15:
You should also mention used sysmodules.

Jboy93, 2020-04-15:
I have a similar issue. Updated hekate to 5.1.4 and atmosphere to 0.11.1 with my switch on 10.0.0 but can't seem to get past the atmosphere boot logo after choosing the fusee primary bin file. All I get is a black screen and nothing is happening.....all my sd files are up to date also the bootloader files etc.

meizano, 2020-04-15:
see "Black screen #883"

RockStar8757, 2020-04-17:
I have the same issues after updating everything, all i get is a black screen after the atmosphere boot logo

fincs, 2020-04-17:
Piracy is not supported.

closed 2020-04-17 by fincs