Atmosphere-NX/Atmosphere #894: New Update not Work pls Help fast

BrosZocken, 2020-04-15:
What's the issue you encountered?
It's the first time for me seeing this bug and I'm using Atmopshere for 3-4 months already. After Atmopshere logo etc i have a black screeen for couple secs and an error.

How can the issue be reproduced?
Downloaded latest 0.11.1 update for Atmosphere, including fusee-primary.bin
Copied every file to SD card, fusee-primary.bin to atmosphere folder.
When asked if I want to overwrite files, I did that.
Plugged in SD card,
Turning on console with RCM loader and jig installed
Crash Report´
A fatal error occurred when running 
Title ID: 010041544d530000
Error Desc: std : :abort() called (0xffe)
Press Power to Reboot
And i reinstalled it

fincs, 2020-04-15:
Delete custom sysmodules and custom themes and try again.

BrosZocken, 2020-04-15:
I reinstalled it on wiidatabase here a link:https://wiidatabase.de/switch-downloads/custom-firmware/atmosphere/

> Delete custom sysmodules and custom themes and try again.

fincs, 2020-04-15:
Delete your sd card's atmosphere folder, redownload/reinstall from [the only official source](https://github.com/Atmosphere-NX/Atmosphere/releases/latest) and try again.

BrosZocken, 2020-04-15:
i do it but it still not work

BrosZocken, 2020-04-15:
> Delete your sd card's atmosphere folder, redownload/reinstall from [the only official source](https://github.com/Atmosphere-NX/Atmosphere/releases/latest) and try again.

I do it but its still not work

fincs, 2020-04-15:
> 010041544d530000

That is a custom sysmodule. I don't think you have deleted the atmosphere folder at all. Make sure it's absolutely identical to the one in the archive.

BrosZocken, 2020-04-15:
Here is a Picture of my contetn **soryy for my bad English** folder [Link to the Picture](https://www.dropbox.com/s/bujwag0qfzmhbbk/Bildschirmfoto%202020-04-15%20um%2020.21.07.png?dl=0)

BrosZocken, 2020-04-15:
see

fincs, 2020-04-15:
Do you get a different error message? Also, macOS is known to corrupt the archive bits on SD cards. You might need to use an external tool in order to repair any potential damage.

BrosZocken, 2020-04-15:
> Do you get a different error message? Also, macOS is known to corrupt the archive bits on SD cards. You might need to use an external tool in order to repair any potential damage.

What is the best external tool?

fincs, 2020-04-15:
IIRC there is an option in hekate to fix archive bits.

BrosZocken, 2020-04-15:
> IIRC there is an option in hekate to fix archive bits.

Ok who i come to the option can you send me a Picture

BrosZocken, 2020-04-15:
> IIRC there is an option in hekate to fix archive bits.
Its still not working i got the same error

BrosZocken, 2020-04-15:
> Do you get a different error message? Also, macOS is known to corrupt the archive bits on SD cards. You might need to use an external tool in order to repair any potential damage.

Fincs pls Help me

BrosZocken, 2020-04-15:
Best man ever thanks

closed 2020-04-15 by BrosZocken