Atmosphere-NX/Atmosphere #895: Black Screen only when booting stock

ollidiemaus, 2020-04-16:
## Bug Report

### What's the issue you encountered?

When I try to boot Stock sysNand using hekate 5.1.4 with this config:
```
[OFW Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
```
First the SEPT Splash Screen appears then I get a black screen.
But when I'm booting sysNand with atmosphere everything works just fine.
Also emuMMC is booting fine as always.

### How can the issue be reproduced?

I just copy paste the atmosphere, sept and switch folder from release 0.11.1 to the SD Card.
Also I copy bootloader from hekate 5.1.4 to the SD Card.
Then I copy the latest fusee-primary in the atmosphere folder.

### Crash Report

There are sadly no Crash Reports.

### System Firmware Version

10.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate v5.1.4
- Official release or unofficial build: 
  - Official release version 0.11.1
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A

### Additional context?
N/A

SciresM, 2020-04-16:
I have no idea what "stock sysnand" means, but I suspect it isn't something that atmosphere actually supports.

ollidiemaus, 2020-04-16:
Usually I could boot without atmosphere just the stock operating system but with the different bootloader to not burn fuses.
Maybe I'm wrong and this is not an atmosphere issue but a bootloader issue then I have to search there...

victormr21, 2020-04-16:
> I have no idea what "stock sysnand" means, but I suspect it isn't something that atmosphere actually supports.

He means HOS in OFW, without Homebrew. I also suffer this problem. I fixed it disabling AutoRCM but I guess that's a temporary solution. And if I'm not wrong CTCaer said that above 7.0.0 you need Exosphere in order to boot Stock sysNAND

River101, 2020-04-16:
Yeah, I'm also experiencing this issue. I returned to the switch after a few months, wiped my sd as exfat and installed the latest atmosphere. Everything works fine. Install switch 10.0 update, reboot, and black screen.

River101, 2020-04-16:
Also, I am unable to boot CFW even with AutoRCM disabled

Kev63, 2020-04-17:
I have the exact same issue. Atmosphere works fine, but not stock original firmware. I downgraded to 9.2.0 with ChoixduJourNX and I will stay in this firmware for now.

River101, 2020-04-17:
> I have the exact same issue. Atmosphere works fine, but not stock original firmware. I downgraded to 9.2.0 with ChoixduJourNX and I will stay in this firmware for now.

Sorry, how did you downgrade as I'd be very interested in that.

Kev63, 2020-04-17:
Using ChoixduJourNX homebrew, but to do so, you have to check if you didn't burn the fuses, otherwise you will brick your Switch. You can check if you burnt them in Hekate, then Console Info, then Fuses, then the orange line Burnt Fuses.

River101, 2020-04-17:
I don't have any orange lines or anything, it just says this:

SKU: 83 - Retail
Sdram ID:0
Burnt fuses: 13/64

RockStar8757, 2020-04-17:
my said this Burnt Fuses (0DM7/6): 13-0. what does it mean ?

shchmue, 2020-04-17:
> Using ChoixduJourNX homebrew, but to do so, you have to check if you didn't burn the fuses, otherwise you will brick your Switch. You can check if you burnt them in Hekate, then Console Info, then Fuses, then the orange line Burnt Fuses.

this is total misinformation. whenever you boot the console via RCM, fuse checks are skipped. the only way to brick your switch like this is by enable autorcm on a PATCHED console, which is blocked by all the current programs used that are able to enable autorcm. if your console is on a firmware below your fuse level, and you boot stock, without RCM, the console will not boot. but you can still boot via RCM

SciresM, 2020-04-18:
This is not an atmosphere problem.

closed 2020-04-18 by SciresM