Atmosphere-NX/Atmosphere #897: Switch will not wake up from sleep mode in Emmu

BruhTh3Lip, 2020-04-16:
## Bug Report

N/A

### What's the issue you encountered?
Switch does not wake up from sleep mode in 0.11.1.

### How can the issue be reproduced?
Load up Atmosphere without any problems but if console goes to sleep pressing power button only produces a black screen and remains stuck.

### Crash Report

N/A
### System Firmware Version

0.11.1

### Environment?

Hekate v 5.1.4 Nyx 0.8.7

hitaloisme, 2020-04-18:
Same issue here.

SciresM, 2020-04-20:
I just made a fresh emummc on 0.11.1, launched using fusee-primary. System settings says 10.0.0|AMS 0.11.1|E.

Sleep/wake-from-sleep works fine.

SciresM, 2020-04-20:
I would double check that you don't have any custom sysmodules or other things installed that are breaking sleep/wake-from-sleep. 

It definitely works fine on stock atmosphere.

closed 2020-04-20 by SciresM