Atmosphere-NX/Atmosphere #901: unable to load atmosphere after boot into emummc

JuicyBerger, 2020-04-17:
Failed To Apply 'nosigchk' Failed to Launch HOS!

### What's the issue you encountered?

After updating to Atmosphere 11.1 I decided to update switch firmware to 10.0, after doing so, When attempting to run emummc I Get the texts 'Failed To Apply 'nosigchk' Failed to Launch HOS!'

### How can the issue be reproduced?
Update to atmosphere 11.1 and update system firmware to 10.0. then attempt to run emummc 
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
1. update atmosphere to latest firmware
2. update system firmware to 10.0
3. attempt to boot emummc
4. all logos come up and switch hard powers off after the atmosphere logo
### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
n/a
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?
n/a
- Additional info about your environment:
- [ Any other information relevant to your issue. ]

fincs, 2020-04-17:
> nosigchk

Piracy is not supported.

closed 2020-04-17 by fincs

jdfalk, 2020-04-23:
Or you could not be a jerk and just answer that the person should remove the kip1patch=nosigchk from hekate_ipl.ini. Which disables piracy and solves his problem, but sure why not be a jerk instead.

JuicyBerger, 2020-04-23:
Awww thank you Johnathan Falk :) I had already figured it out a couple
hours later but thank you for actually coming to a helpful conclusion
you’re so sweet :3

deadair3210, 2020-04-23:
The whole point of something not getting supported is it doesn't get answered or talked about......