Atmosphere-NX/Atmosphere #903: I need help, I updated my console official firmware to 10.0.0, am i able to set up a fresh atmosphere now?

bonjorys, 2020-04-17:
Basically, 
I updated my official firmware to latest 10.0.0, now im trying to set up a fresh atmosphere install from scratch (am i even able to do this at my firmware level?) and keep getting an error when i run tegraexplorer through tegrarcmgui, biskey dump error 57, I am able to continue past this point and install everything but when i try to launch atmosphere im just getting a black screen, 
however i still have a working copy of atmosphere on a different micro sd card at firmware 9.2.0 
I had recently bought a bigger micro sd card which is why im trying to set up a new one, I am also considering using Ease partition manager to clone my working card to the bigger one as i believe that works also, 
Any help or info would be much appreciated, thanks!

RockStar8757, 2020-04-17:
The issues is not with atmosphere, the issues is Joonie86 Kosmos patches. I got the same issues after updating everything to the newest. i also got the black screen, but when i remove Joonie86 Kosmos patches the black screen is gone but couldnt't play any games. we gotta wait for an update patches from jooni86 and everything should be working fine with atmosphere.
I don't think you can use a different mico sd card unless you downgrade 10.0.0 back to 9.2.0 than you could use it.
you can remove atmosphere off your SD card and fresh install a new copy. you would see what am talking about. hope this help.

fincs, 2020-04-17:
Piracy is not supported.

closed 2020-04-17 by fincs