Atmosphere-NX/Atmosphere #905: Error 2001-0125 when launching hbmenu via app override on 5.0.2

DeadCakeTKL, 2020-04-17:
## Bug Report

### What's the issue you encountered?

Atmosphere 0.11.1 does not let you boot into hbmenu via app override on 5.0.2. Applet mode works via album.

### How can the issue be reproduced?

- Obtain EmuNAND running on 5.0.2
- Install a fresh copy of Atmosphere 0.11.1
- Boot into EmuNAND using Hekate
- Navigate to Home Menu and open an app while holding down configured app override button
- Selected app opens and hangs on loading screen for 20+ seconds
- Error is produced

### Crash Report

[01567493967_0100000000000023](https://gist.github.com/DeadCakeTKL/d167c7e38725d45fc188958c661fc25e)

### System Firmware Version

5.0.2 EmuNAND
9.1.0 SysNAND

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
  - 0.11.1 Official
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/A

### Additional context?

Installing a fresh copy of Atmosphere 0.10.x does not produce the same error and functions as intended.

SciresM, 2020-04-20:
This was a hbl error and not an atmosphere one. https://github.com/switchbrew/libnx/pull/405 has been made to fix it, should be fixed in the next atmosphere release.

closed 2020-04-20 by SciresM

mitchellmebane, 2020-09-20:
Did this fix ever get released?  I'm seeing the same behavior on 4.0.1 with Atmosphere 0.14.4.  Should I open a new issue?