Atmosphere-NX/Atmosphere #906: Black screen after Ams logo

UdoChudo, 2020-04-17:
## Bug Report

### What's the issue you encountered?
I got Lexar 256gb new SD card, after cloning Old SD card to New SD card, i can't boot to CFW from New SD. 
what i'm tryed
1) Format to fat32
2) Format exFat
3) Boot through Hekate
4) Boot through fusee-primary

### How can the issue be reproduced?
Got Lexar 633x 256gb
Cloning old SD card to new with resizing Fat32 first partition

### System Firmware Version

9.2.0</br>
Ams 10.5/11.1 (tried both)
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: 
Booting from Hekate 5.1.3, 5.1.4
- Official release or unofficial build: 
Ams 10.5, Ams 11.1
- Do you have additional kips or sysmodules you're loading:
emummc driver

### Additional context?
When i push payload fusee-primary.bin i got:
Sept splash screen
Ams on blue background
Black screen w/o backlight

When i boot using Hekate fss0=fusee-secondary.bin
Customboot splash
Sept logo
Black screen w/o backlight

Old SD card was Samsung Evo plus 128gb and on it, same config AMS/Hekate/FW working good
I’ll try to explain everything from the beginning. I cloned my old SD card to a new, larger one. After that, when I try to boot into the EmuMMC after the Atmosphere logo, I get a black screen. Likewise, if I try to boot into the System Nand, the exact same card from which a full copy was taken sector by sector works with the config attached above and everything works fine, but on the new card loading stops after the first splash screen of the atmosphere

UdoChudo, 2020-04-18:
Update news,
After clear EmuMMC creating Ams wont boot to SysCFW/EmuCFW
Reproduce:
Formating whole SD by Rufus to Fat32
cut 30Gb for Emu stay unformated
Place downloaded Ams files to SD
Push fusee-primary.bin
Booting to SysCFW
After this i'm create EmuMMC by Nyx 
Push fusee primary and now i can't boot not to Sys and Emu CFW

Anti-Apple4life, 2020-04-18:
I'm having the same problem after updating to 10.0.0 exfat. I tried using both exfat and fat32 formats and I got a totally black screen. After updating the Sept and Atmosphere folders, I got the Sept splash screen and the default Atmosphere splash screen (shown below), but after that, I got a black screen. I ran Switchroot Android on another card and it loaded and worked perfectly. I think a new release should be made.
[image]

SciresM, 2020-04-18:
Sounds like sd card issues or badly formatted emummc. Either way talk to @ctcaer.

closed 2020-04-18 by SciresM

UdoChudo, 2020-04-18:
Already talk with CTCaer, emummc is normal and formating is ok too. He say what it's not hekate problem

UdoChudo, 2020-04-18:
> Sounds like sd card issues or badly formatted emummc. Either way talk to @CTCaer.

i'm pass h2test, formating card with Rufus after creating emunand with Nyx, All checks complete without issues

SciresM, 2020-04-20:
I am literally entirely confident this is not an atmosphere issue. This kind of issue only happens when there's extra kips in the kips folder fucking things up. 

Delete your atmosphere folder, re-extract latest update, do not add anything else to your folder. It will boot.