Atmosphere-NX/Atmosphere #908: Re-authorize "development" applications execution on retail units

OperationNT414C, 2020-04-18:
## Feature Request

### What feature are you suggesting?
#### Overview:

Since Atmosphere version 0.11.1, on a retail unit, some "development" applications cannot work. Would it be possible to add a setting to make them work again?

#### Smaller Details: 

Following changes in file "ldr_meta.cpp", function "ValidateAcid", prevent "development" application run:

```
            /* Validate that the acid is for production if not development. */
            if (!IsDevelopmentForAcidProductionCheck()) {
                R_UNLESS((acid->flags & Acid::AcidFlag_Production) != 0, ResultInvalidMeta());
            }
```

I understand that this is a very specific case that most people won't need. So, it could be activated under an option in "system_settings.ini".

#### Nature of Request:  

Avoid a regression/limitation which was not there on Atmosphere v0.10.5

### What component do you feel this would best fit within?

Stratosphère - Module "loader.kip"

### Why would this feature be useful?

Use a specific "development" application to backup/restore savedata, have controller details (such as their internal firmware version), manage user accounts...

deadair3210, 2020-04-18:
this still has the issue of allowing illicitly used SDK software

OperationNT414C, 2020-04-18:
I don't see what you are talking about...
Are you refering another locked issue? :-)

deadair3210, 2020-04-18:
You opened a issue 2 days ago about trying to get DevMenu to work again, which is pretty much the only dev software that people ever want to run on the switch.

fincs, 2020-04-18:
Closing and locking.

closed 2020-04-18 by fincs