Atmosphere-NX/Atmosphere #909: Crash when trying to re-initialize the system

igoticecream, 2020-04-18:
## Bug Report

### What's the issue you encountered?

System crash after hitting the red button to re-initialize the console via system settings

### How can the issue be reproduced?

1. Boot into emummc of HOS 10.0.0 and Atmopshere 0.11.1
2. Re-initialize the console via system settings

### Crash Report

No crash report was generated after several crashes but here's some info i see on screen:
title ID 010000000000001f, error code 0x25A0B (2011-0301). If you need more info i can post a photo of the screen, but sadly no report is being generated on the sd

### System Firmware Version

10.0.0 | 0.11.1 | E

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Chainload hekate -> fusee
- Official release or unofficial build: Official 0.11.1
- Do you have additional kips or sysmodules you're loading: None, stock
- Homebrew software installed: None, stock

### Additional context?

- Additional info about your environment: Fresh emummc partition that i try to re-initialize with clean atmosphere

SciresM, 2020-04-20:
Reproduced. It only affects 10.0.0, NS is failing somewhere using new fs::OpenBisWiper code.

Will investigate to fix.

SciresM, 2020-04-20:
Issue occurs even with fs.mitm disabled, so doesn't seem to be a result of fs.mitm intercepting NS FS-calls incorrectly.

Mathieu14, 2020-04-24:
I don't know if it will help you to find the bug but i had exactly the same error (0x25A0B) after i made a factory reset with choixdujour when upgrading to 10.0.0.1. I recreated the emummc, copyed atmosphere and this bug happened. I used a save with atmosphere 9 to go around the bug.

hexkyz, 2020-04-25:
@Mathieu14 That does indeed give some hints as to what might be happening, thank you! This bug is directly related to the changes introduced in firmware 10.0.0 which changed how FS performs factory resets (https://switchbrew.org/wiki/Filesystem_services#OpenBisWiper).
I'm in the process of reverse engineering the issue and will report back when solved.

SciresM, 2020-04-28:
Solved by 491ba8fdcfd39a503bedd21b282991fc19aec7d4

closed 2020-04-28 by SciresM