Atmosphere-NX/Atmosphere #914: Cannot open HBMenu in applet mode whild running a title

OldGnashburg, 2020-04-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ While running a title such as Bayonetta and then launch the Album applet to get it the HB Menu to launch Edizon the Switch crashes with yellow text saying:
Panic occurred while running Atmosphere

Title ID: 010041544D530000
Error Desc: std::abort() (0xFFE)]</br>
[ Just System modules like Tesla and sigpatches for using Homebrew NSP's built with NRO2NSP ]</br>
[ I have Tesla, Fizeau, ldm_mitm, Sys-clk, Sys-tune, SaltyNX and a bunch of overlays installed but I have turned off Fizeau, and ldn_mitm. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I launch Bayonetta, make sure I'm in game, and press the HOME Button, move to and press A over the Album Applet. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official Atmosphere 0.11.1 EmuNAND Launched from Hekate]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Not sure if this is a memory related issue because of System modules,]

OldGnashburg, 2020-04-21:
[report_03238780e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4507651/report_03238780e.zip)

fincs, 2020-04-21:
> sigpatches for using Homebrew NSP's built with NRO2NSP

Piracy is not supported. The only supported way to run homebrew is through nx-hbloader processes.

> I have Tesla, Fizeau, ldm_mitm, Sys-clk, Sys-tune, SaltyNX and a bunch of overlays installed but I have turned off Fizeau, and ldn_mitm.

As usual, custom system modules depleting system resources (especially memory) and causing random crashes.

closed 2020-04-21 by fincs

basvijfwinkel-cinarra, 2020-04-29:
This issue also happens with legally purchased games.
I deleted sys-clk and sys-con and then the issue was solved.