Atmosphere-NX/Atmosphere #915: Opening homebrew menu from album or games gives me

DitDitGitHub, 2020-04-22:
Hi all,
After updating multiple apps via the Homebrew App Store, whenever I open the homebrew menu it gives me a fatal error which crashes my Switch. This is the report: 
[71792941139_01005de00d05c000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4518021/71792941139_01005de00d05c000.log)
All help is appreciated.

JKakaofanatiker, 2020-04-22:
Same here. R on startup doesn't work either.

SciresM, 2020-04-22:
1) Use the issue templates, please. 

2) Does the error occur if you delete atmosphere/hbl.nsp, /hbmenu.nro and replace them with the ones in the latest release zip?

DitDitGitHub, 2020-04-22:
> * Use the issue templates, please.
> * Does the error occur if you delete atmosphere/hbl.nsp, /hbmenu.nro and replace them with the ones in the latest release zip?

I didn’t get a template... sorry. I will try to delete the two files. Thanks!

DitDitGitHub, 2020-04-22:
Still not working... have updated both files. When I open the menu, it stays open for a couple of seconds before crashing.

SciresM, 2020-04-22:
@DitDitGitHub The crash report you posted is corrupted. Can you see if there are any other crash reports which aren't corrupted?

DitDitGitHub, 2020-04-22:
> @DitDitGitHub The crash report you posted is corrupted. Can you see if there are any other crash reports which aren't corrupted?

These are the others... I'm not sure if they are corrupted or not.
[77018863778_01005de00d05c000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4518466/77018863778_01005de00d05c000.log)
[13266312791_01005de00d05c000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4518467/13266312791_01005de00d05c000.log)
I think they are the same error...

masagrator, 2020-04-22:
Both are corrupted. Have you tried to use different sdcard? Are you using exFAT or FAT32? Emummc or sysnand? 
Etc

DitDitGitHub, 2020-04-22:
> Both are corrupted. Have you tried to use different sdcard? Are you using exFAT or FAT32? Emummc or sysnand?
> Etc
I haven't used a different sdcard, I'm on Fat32 and Sysnand. Just had an idea, going to try removing all hbmenu apps. It started once I updated them so it might be one of them? I have no idea what I'm doing lol

DitDitGitHub, 2020-04-22:
So after deleting all the apps I get a new error:
[00527960916_430000000000000b.log](https://github.com/Atmosphere-NX/Atmosphere/files/4518584/00527960916_430000000000000b.log)

DitDitGitHub, 2020-04-22:
And I get the same error while opening some games

masagrator, 2020-04-22:
Sys-botbase is crashing. Delete it.

DitDitGitHub, 2020-04-22:
Thank you! The problem is fixed!

closed 2020-04-22 by DitDitGitHub