Atmosphere-NX/Atmosphere #917: Crash when accessing Album to load into HBMenu

HairyMilkshakes, 2020-04-23:
Getting the following error code when trying to launch into the HBMenu from album icon: Error Code: 2345-0033 (0x4359)

All files are obtained from sdsetup.com and so I would assume they are all updated. Firmware is 9.2.0 Atmosphere 0.11.1

masagrator, 2020-04-23:
https://github.com/Atmosphere-NX/Atmosphere/issues/new?template=bug_report.md

Template

HairyMilkshakes, 2020-04-23:
Thanks, didn't see that before. It's not generating a crash report inside the directory in Atmosphere, for some odd reason.

HairyMilkshakes, 2020-04-23:
I deleted the files inside atmosphere/contents and replaced them with the defaults from sdsetup and it appeared to function again. not sure which module is crashing and will advise.

fincs, 2020-04-23:
"sdsetup"

Care to elaborate?

deadair3210, 2020-04-23:
It's a website where you can select items and it will build the SD card folder for you

HairyMilkshakes, 2020-04-23:
Resolved. No longer an issue after I did this: https://github.com/Atmosphere-NX/Atmosphere/issues/917#issuecomment-618208140

closed 2020-04-23 by HairyMilkshakes

kkirby, 2020-04-28:
I think the issue is with sys-con. Just a heads up for anyone else having issues.