Atmosphere-NX/Atmosphere #919: Error when i try to compile Atmosphere

aldone1, 2020-04-23:
This is the error where is the problem? Anyone can help me?
**/opt/devkitpro/Atmosphere-master/exosphere/src/configitem.c: In function 'configitem_get':
<command-line>: error: invalid suffix "x" on integer constant
<command-line>: note: in definition of macro 'ATMOSPHERE_RELEASE_VERSION_HASH'
/opt/devkitpro/devkitA64/base_rules:18: recipe for target 'configitem.o' failed
make[1]: *** [configitem.o] Error 1
Makefile:130: recipe for target 'build' failed
make: *** [build] Error 2**

HookedBehemoth, 2020-04-23:
That constant is defined in [here](https://github.com/Atmosphere-NX/Atmosphere/blob/master/exosphere/Makefile) based on the git hash

You'll need to compile on a git repo. Downloading the zip won't do.

aldone1, 2020-04-23:
Ok, thank you very much

closed 2020-04-23 by aldone1