Atmosphere-NX/Atmosphere #920: Help

user466, 2020-04-23:
So I'm running Atmosphere 0.11.1 and my switch updated through the system settings to 10.0.1 and now it doesn't want to boot up past the sept, and I've already deleted the atmosphere, Sept, and and homebrew file and reinstalled them again but it still doesnt boot up.

astroavenger9, 2020-04-23:
i'm having the same problem. my switch updated to 10.0.1 and now it won't boot past the sept screen. also made sure that i got the latest verson of atmosphere.

user466, 2020-04-23:
UPDATE: So I booted in through normal and it says I need a system update to use the MicroSD but my system is already up to date

deadair3210, 2020-04-23:
You are missing the exfat update likely

astroavenger9, 2020-04-23:
@user466 i did that too and said i needed to update as well. but when i click on update it says i have the latest version.

@deadair3210 what do you mean?

deadair3210, 2020-04-23:
If your SD card is formatted to exfat, you require a small update to install the driver so the switch can read the card. If your SD card is formatted to exfat, or for some reason the switch thinks it is, and you don't have that update, the switch will say that you need a update to read the card

fincs, 2020-04-23:
There's something really odd going on here, because if your card was formatted as exFAT but your firmware is the FAT32-only variant, Atmosphère would not have been able to boot at all, since Atmosphère relies on accessing the SD card filesystem from very early on in the Horizon OS bootup sequence. Please make sure you have Atmosphère correctly installed (and use fusée instead of e.g. hekate to verify that you are not in fact booting up to stock firmware); as the expected result should have been a boot failure.

astroavenger9, 2020-04-23:
@deadair3210 my sd card is indeed formatted to exfat. my switch does say it need to be updated(when booted normaly) but when i click on update it says there nothing to update.

@fincs as far as i know i have atmosphere correctly installed. i'll aslo note that my switch was booting up just fine before i installed the 10.0.1 update and as soon as i updated it stoped booting up all the way.

astroavenger9, 2020-04-23:
would reformatting my sd card fix the problem because this sounds like a formating problem. don't really want though because i got a lot of games on there and its going to take forever to back everthing up and then put back on.

fincs, 2020-04-23:
Since you have exFAT, it's very possible that you have filesystem corruption. You might want to get that SD card reformatted as FAT32 anyway.

astroavenger9, 2020-04-23:
@fincs sigh i'll try that then

SciresM, 2020-04-24:
Closing, reopen with issue templates if the error occurs post reformat.

closed 2020-04-24 by SciresM

astroavenger9, 2020-04-24:
ok once i backed everything up and tried to format it FAT32 wasn't availble. so i also tried booting up the switch normally without atmosphere and getting my switch to format my sd card. when i did it said i needed a system update. so when i clicked update it said i already got the latest update. so i'm kind of stuck. also i've been playing on my switch with my sd card in exfat format for months with it working just fine. why am i haveing problems now. i did try formatting my sd card keeping it in exfat  and installing atmosphere on agian and it still doesn't booot up all the way. it show the booting screens but thats it. what am i doing wrong.

fincs, 2020-04-24:
Try using Rufus to format your SD card; that lets you format big cards as FAT32 even if Windows doesn't let you.

astroavenger9, 2020-04-24:
hhhhggh