Atmosphere-NX/Atmosphere #921: Mount XCI files

Kratos-Mundus, 2020-04-23:
## Feature Request

### What feature are you suggesting?
#### Overview:
- A feature that make it possible to mount XCI files</br>

#### Nature of Request:  
- Change

### What component do you feel this would best fit within?

- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>

- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
- This feature would be useful because you dont have to Install NSP files every time you want to play a game because sometimes it can take a lot of time to install a NSP file</br>

SciresM, 2020-04-23:
Atmosphere doesn't support the installation of content except through the eShop.

Thank you for using the issue templates.

closed 2020-04-23 by SciresM

ThatNerdyPikachu, 2020-04-23:
Atmosphere does not support piracy.

Kratos-Mundus, 2020-04-23:
So it is impossible to load XCI files via Atmosphere?
Or is there a way to do that via Atmosphere

fincs, 2020-04-23:
Piracy is not supported, see previous comment.