Atmosphere-NX/Atmosphere #923: fatal error occured when running atmosphere (titleid 0100000000000005)

NarutoRenegado, 2020-04-24:
## Bug Report

### What's the issue you encountered?

[Update my sysnand to 10.0.1 and atmosphere, but when injecting the payload from tegrarcm the atmosphere logo screen appears and then I get this error. ]</br>
[ a fatal error occured when running atmosphere
title id: 0100000000000005
error desc: std::abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_00000000011a266a.bin]</br>

### How can the issue be reproduced?

[ Update the sysnand to 10.0.1, I also updated atmosphere to 0.11.1, when injecting the fusee-primary.bin with tegra RCM the screen with the Atmosphere logo appears and then a black screen with the error described above]</br>

### Crash Report

[ The content of the fatal_error file (as it is created as named in the error message) looks like a binary format. I'm not sure how to paste this into a gist. Is there any tool to read the contents of this bin-file in a human readable way? Or should I upload the bin-file anywhere else?. ]

### System Firmware Version

10.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build:  0.11.1
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ Checkpoint and Edizon ]

### Additional context?

- Additional info about your environment: The micro sd is formatted in exfat, i already tried to format the micro sd memory again, but the error persists, also try fat32

SciresM, 2020-04-24:
Bin file upload would be great, thank you.

hdnes, 2020-04-24:
Seeing almost the exact same issue: Title ID is different but same idea.

SciresM, 2020-04-24:
"title id is different" --> then it is a completely different, unrelated issue, and you should send me the relevant fatal error binary.

git2gether, 2020-04-24:
same error.  is there a specific fusee-primary.bin file we should be using as payload?

AndreasDay, 2020-04-24:
Same issue here 
Tried a new SD Card, reinstalling, exfat and fat32, same issue 
Report uploaded here 
http://www.mediafire.com/file/wlpk9b3pc88csxi/report_0000000001064e1b.bin/file

SciresM, 2020-04-28:
I am pretty sure this is resolved by 0.12.0 (incidentally, due to kernel physrand fixes), relevant report was triggered by boot sysmodule not having mmio mapped it is guaranteed to have mapped.

If this is triggered on 0.12.0, open a new report and send me the binary.

closed 2020-04-28 by SciresM

drminh2807, 2020-05-02:
Same issue
I'm using 0.12.0 on system version 10.0.2
My crash report 
[report_0000000005a9edba.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4567137/report_0000000005a9edba.bin.zip)

SciresM, 2020-05-02:
@drminh2807 your report is completely unrelated to this issue. You have archive bits set on the atmosphere folder, run the hekate tool to unset them.

insomiagamerz, 2020-05-13:
Hi @SciresM 
I hit the same exception when Lauch  and select CDW - sysMMC from hekate v5.2.1.
Stock - sysMMC can launch to original nintendo firmware 10.0.2
Another issue is when launch Payloads -> fusee-primary,bin, atmosphere would not launch (Black screen with backlight in on). This happend after I created enuMMC (SD File).
Before create enuMMC (SD File).
1. I can still launch Payloads -> fusee-primary,bin. 
2. Lauch  and select CDW - sysMMC will show different exception (I cannot remember the exact word but its in Yellow color)
3. I can turn on switch without RCM loader (power button) and it will directly load from original nintendo firmware 10.0.2 which now it would not turn on at all.

Appreciate anyone who can help to solve this issue.
The crash report:
[https://www.mediafire.com/file/kfobmx51cgrczro/report_000000000135d43d.zip/file](url)

iixela, 2020-07-01:
Hi!
I have the sale error. Update from 9.0.1  to 10.0.4  and i can load payloads (heckate, primary or reinx) but show black screen after.

Thanks a lot for you support! 

Fatal error reports:

http://www.mediafire.com/file/timvnwnpt6rq1ly/report_0000000001c91b68.bin/file

http://www.mediafire.com/file/xg07i0yjmlaeapa/report_000000000128b2f7.bin/file

http://www.mediafire.com/file/uh8dmzubnr8n2e7/report_0000000000c3933a.bin/file

MasterOrg012, 2024-09-05:
the same issue here its the folder whit the reports [https://www.mediafire.com/folder/kgmt81rieuc9m/fatal_errors](url)