Atmosphere-NX/Atmosphere #924: Fatal error (Bad SVC)

UnderRev0lved, 2020-04-24:
## Bug Report

### What's the issue you encountered?

I didn't change ams and used vanilla from release on Github.
While playing "PC Building Simulator", a fatal error screen appeared with "Bad SVC" error

### How can the issue be reproduced?

Playing PC building simulator and going fast in step.

### Crash Report

### System Firmware Version

9.2.0</br>
</br>
</br>
</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Fusée
- Official release or unofficial build: 
Official 0.11.1
 
- Do you have additional kips or sysmodules you're loading:
Not sure
- Homebrew software installed: 
No

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-04-24:
Please upload the relevant crash report...?

UnderRev0lved, 2020-04-24:
Can't find it

UnderRev0lved, 2020-04-24:
I'm searching it

SciresM, 2020-04-24:
Was it an atmosphere fatal error (black screen, report dumped to fatal errors dir) or a game fatal (e.g. ams fatal screen displayed by fatal sysmodule?)

Also can't really do anything without knowing what program id it occurred in lol

UnderRev0lved, 2020-04-29:
[report_000000068bfad0de.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4552870/report_000000068bfad0de.zip)
Error report of Bad SVC 0x301 while playing FUZE4

UnderRev0lved, 2020-05-01:
As seen on DIscord, the problem is from the switch itself.
Issue closed

closed 2020-05-01 by UnderRev0lved

Thebrianzard, 2023-08-13:
Hello Ghost, how did you fix this issue?