Atmosphere-NX/Atmosphere #925: Black screen after updating to 0.12.0

idmew, 2020-04-25:
Hi, I just updated to the latest release and I'm getting a black screen after the Atmosphere logo.
Everything was fine before and I never had any problems updating.
Was on 0.11.1 and 9.2 firmware.
Can someone help me please?

i4s33u, 2020-04-25:
I have this issue too :(

daniele728, 2020-04-25:
Same issue here, FW 9.2 , previous atmo 10.4.
Tried to turn back with 10.4, black screen after atmo logo screen.

idmew, 2020-04-25:
I was able to make it work by deleting the Atmosphere folder entirely and copy the new one over.

AbuYo, 2020-04-25:
I tried deleting Atmosphere folder and copying the new one over too and it successfully booted up, but I could not run any installed applications, it always shows the error "Could not start the software"

idmew, 2020-04-25:
> I tried deleting Atmosphere folder and copying the new one over too and it successfully booted up, but I could not run any installed applications, it always shows the error "Could not start the software"

Same thing happened to me, you have to add the patches from here: [https://gbatemp.net/threads/atmosphere-patches-es-fs-and-loader-fusee_primary-only.562812/](url).

After I did that everything went back to normal.

fincs, 2020-04-25:
Piracy is not supported.

closed 2020-04-28 by SciresM