Atmosphere-NX/Atmosphere #926: docs/cheats.md has not been updated to reflect the more recent opcodes.

BtEO, 2020-04-25:
## Feature Request

### What feature are you suggesting?
#### Overview:
`docs/cheats.md` does not currently reflect the complete set of opcodes available. In particular code types C1, C2, C3, FF0, FF1</br>

#### Nature of Request:  
- Change

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
It would be a lot easier for cheat creators than figuring out how the new opcodes work from the source.</br>

hexkyz, 2020-04-25:
Thank you for reporting! The documentation is currently being re-written entirely. I'm assigning this issue to myself and will close at the same time as #797.

hexkyz, 2020-05-01:
Closing this issue as the documentation has been fully updated.

closed 2020-05-01 by hexkyz