Atmosphere-NX/Atmosphere #927: Multiple "contents" folders: one for system titles, one for user-provided titles

natinusala, 2020-04-25:
What would you think about having two `contents` folders:
- one containing the "system" titles currently bundled in the releases
- another containing the user-provided titles only (would be empty or left out in release archives)

That would:
- allow knowing which titles are installed versus which are Atmos reimplementations
- prevent accidental deletion of "system" titles
- make it easy to reset the Atmos install (just empty out the user provided titles folder)

The use case here is someone installed too many sysmodules (preventing the console from booting) and wanted to remove all of them. In the end they had to remove the contents folder and put back the files from the Atmos release.

Clydefrosch22, 2020-04-25:
i suggested something similar a while ago, but with subfolders inside contents here https://github.com/Atmosphere-NX/Atmosphere/issues/870
Definitely support anything that helps manage that folder.

UnderRev0lved, 2020-04-29:
That's what I exactly thought.
That would be a welcome improvement

misson20000, 2020-04-29:
Reminds me of how systemd has separate directories for package-provided unit files and hand-written unit files. On that note though, dealing with multiple software distributions that all place files in a common directory tree smells a lot like a general package management problem.

Separating AMS content from user content is a very good start, but I would love to see package management be like what we've done for exefs_patches, where they are grouped by patchset: a `packages/` directory where each directory represents a different software distribution and has its own `contents/`, `exefs_patches/`, etc. directories. Perhaps even specify a manifest with information like human-readable package name, author, release date, homepage, license, supported AMS versions, supported HOS versions, etc., not so much for AMS to use as to create a standard for homebrew package managers to use.

- atmosphere/
  - contents/
    - (AMS distribution contents)
  - kip_patches/
    - (AMS distribution kip patches)
- packages/
  - my_epic_botw_mod/
    - (AMS directory tree format replicated here)
    - contents/
      - 01007EF00011E000/
        - (...)
    - exefs_patches/
      - (patchset name optional here, since it's kinda redundant with this format)
      - 2094F152836683A3632C6153FD84AB80.ips
    - manifest.ini
  - some_sysmodule
    - flags/
      - disable.flag
    - contents/
      - 0100000aabb00001/
        - exefs.nsp
        - flags/
          - boot2.flag
    - manifest.ini
  - sysmodule_with_multiple_titles_like_twili
    - contents/
      - 0100000000006480/
      - 0100000000006482/
    - manifest.ini

HookedBehemoth, 2020-04-29:
> supported AMS versions, supported HOS versions
I'd love to have this feature.

The manifest could also be expanded to e.g. emuMMC on/off, Calcio/Hoag/...

SciresM, 2020-04-29:
I am mostly okay with this conceptually.

cheesycod, 2020-06-03:
I like the idea, however this could be more confusing to the users. Also, we can't easily migrate existing installs to this

natinusala, 2020-06-03:
We don't need to migrate anything since the current folder will still work, we'll just need to encourage users to move their stuff to the new folder

fennectech, 2020-10-27:
That will be the first thing i do if/when this gets implemented   Not having to pick through the contents folder to delete sysmodules without deleting mods would be a godsend.

ELY3M, 2020-11-07:
I agree with this idea.

fennectech, 2020-11-07:
The ability to add names after the folder title IDs would also be incredibly nice

SciresM, 2020-11-07:
No matter what solution we take regarding multi-contents-folder support, atmosphere is never going to allow non-program id folder names (or data after the folder). The complexity overhead is flat out not worth it, it introduces entirely new classes of error case bullshit (like two or more folders with the same program id). 

It turns a simple one way mapping via string format into an ambiguous operation that requires a full directory scan.

Not happening.

fennectech, 2020-11-07:
Makes alot of sense.  Ill jsut have to stick to my text file index solution

victorgneves, 2021-01-17:
Now I came to Issues to open a post exactly like this.
I find it very bad in every update of Atmos to have to keep looking at which folders I need to delete and which ones I don't need before updating.
My suggestion would be a file /atmosphere/"contents.ini" that the user edits and creates the corresponding folder.
All of them with the same functionality.

Inside the file ...

[contents]
/ atmosphere / contents
/ atmosphere / (whatever I want)

example:

[contents]
/ atmosphere / contents
/ atmosphere / Mods
/ atmosphere / Translations

Maybe come as standard ..

[contents]
/ atmosphere / atmos_contents
/ atmosphere / user_contents

Another suggestion would be not to create a "contents.ini", but to add the function in /atmosphere/config/system_settings.ini

Sorry for any mistake, I'm using Google Translator

SciresM, 2021-01-17:
I am emphatically not going to introduce ambiguity into what folder is searched for contents.

There are serious design issues with splitting the folders, it's not a simple thing to do. There have been many developer discussions in the ReSwitched discord about this ^^

SciresM, 2021-01-17:
I'm still interested, to be clear, in some kind of separation. But the current thought is to package the atmosphere sysmodules into a "stratosphere.romfs" or something.

victorgneves, 2021-01-17:
Ok, thanks for explaining.
Anyway, they are only small suggestions, your work is excellent.

fennectech, 2021-01-17:
Separating sysmodules from atmo from game mods would be nice.

closed 2021-03-21 by SciresM

fennectech, 2021-03-22:
This doesnt exactly recreate the functionality requested here. It doesnt allow us to separate game mods from user installed sysmodules for easy backup of mods without backing up user sysmodules.   A separate contents folder tor romfs mods that separates from user modules would make it much easier for working with lots of mods.   Picking mods from modules is a pain. Even without atmosphere modules being in there.     It is really nice tho and makes the process easier.

SciresM, 2021-03-22:
@fennectech Actually, it completely recreates the functionality requested here.

The OP is

```
* one containing the "system" titles currently bundled in the releases
* another containing the user-provided titles only (would be empty or left out in release archives)
```.

The system titles bundled in the release and the user-provided titles are now separate. This issue makes zero mention of separating mods from user sysmodules.

Separation of mods from user-sysmodules is in any case less *clearly* desirable and isn't actively planned.

fennectech, 2021-03-22:
How hard would a third be.   I thought i mentioned it

PR #1408 (2021-03-21): ams: distribute sysmodules in single file as stratosphere.romfs
After this PR is merged, atmosphere will no longer use the /contents/ folder for its system modules.

This PR will include a mechanism for cleaning up pre-ams-0.19.0 contents, which can be removed further in the future.

This closes #927, and is critical to ever tackling #704.