Atmosphere-NX/Atmosphere #936: Error 20001-0123 (0xf601)

PauchardThomas, 2020-04-29:
## Bug Report
Hi,

### What's the issue you encountered?

Atmosphere crash after playing <1min at PKMN Sword. However, there is no crash on Mario Kart for instance. Maybe a RAM , memory issue ??

### How can the issue be reproduced?

* Obtain EmuNAND running on 8.0.1
* Install a fresh copy of Atmosphere 0.11.1-master-d81a3bdc
* Boot into EmuNAND using Hekate
* Launching Pokemon sword
* Wait around 1min
* Error is produced (Error 20001-0123 (0xf601))

### Crash Report

Got 2 crash reports at same time :
 * [report 1](https://gist.github.com/PauchardThomas/8db44fe17d77a32f4f85d243cb717d5c)
 * [report 2](https://gist.github.com/PauchardThomas/95125af7b67366c44ead30dabd11208d)

### System Firmware Version

* Sysnand 8.0.1
* Emunand 8.0.1|AMS 0.11.1|E</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate v5.1.4
- Homebrew software installed: [ * ]

### Additional context?

N/A

SciresM, 2020-04-29:
This is caused by (outdated(?)) ldn.mitm, remove it.

closed 2020-04-29 by SciresM

PauchardThomas, 2020-04-29:
Hi @SciresM  can you be more expressive ? How to remove this "ldn.mitm" ? Thanks

SciresM, 2020-04-29:
You installed a custom system module. To remove it, remove the atmosphere/contents/4200000000000010/ folder.

PauchardThomas, 2020-04-29:
Thank you so much