Atmosphere-NX/Atmosphere #937: Atmosphere should protect against custom modules.

Nazosan, 2020-04-29:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Protect against apps installing possibly hostile custom modules (especially potentially without user input.)

#### Smaller Details: 
- Protect certain Atmosphere folders and configuration files against modification while running (with simple on/off toggle in configuration file on by default.)

#### Nature of Request:  
So a while back I tried the Tinfoil.io tinfoil since it supports some newer formats among other things. I was a bit irritated at some of its actions including something that could very well be a potential trojan (if not now then later) or, if nothing else, just plain something that has potential to go wrong.  Regardless, I'm not saying that one is definitely bad (I would assume it's not or people would be speaking up by now) but it does highlight a potential security issue all around. Pretty much any app could ultimately just install its own custom modules and they don't really even have to ask you then those modules can be used to bypass all protection mechanisms built-in to Atmosphere anyway.  That means hostile apps can do it too.

I know it isn't Atmosphere's job to hold the user's hand and act as a virus scanner or whatever, but it does seem like allowing anything to just drop in custom modules at will is a recipe for disaster all the same.

It probably also needs to protect its INI files from modification while it's running I think. I realize this could be annoying in regards to FTPing changes over, but it's a small price to pay given that an app can currently just disable all protections anyway.

### What component do you feel this would best fit within?
I'm not exactly sure which part handles the currently existing protection mechanisms.  I would guess the security monitor, Exosphère, but I don't know enough of its workings to say.

### Why would this feature be useful?
Since apps can use custom modules or INI changes to bypass Atmosphere's existing protections, it defeats the purpose as soon as app creators realize this fact (as at least one already has.)  These protections exist for a reason and it isn't all that hard for a user to disable them if they legitimately and knowingly need to, but it defeats the whole purpose if an app can just bypass or disable them itself.

SciresM, 2020-04-29:
"Tinfoil.io tinfoil since it supports some newer formats among other things"

To be clear, Atmosphere doesn't support warez/warez tools.

That aside, I'm not sure that this is even really possible. There are a lot of legitimate use cases that involve the installation/management of custom system modules. It would be impossible to update atmosphere, for example, if content management was blocked wholesale, and it's unclear to me how you allow mod managers if you don't allow on-the-fly content folder edits.

It kind of seems to me like the best solution is to prevent anything catastrophic from occurring (this is the bis protection policy in general), and then ask people to only run homebrew from sources they trust?

Nazosan, 2020-04-29:
Oh I'm not the biggest fan of Tinfoil.io and I don't use it.  Mostly it seemed nice since it supports compression.  But the real point wasn't "I want to use Tinfoil.io" -- if I did I'd just let it install its custom module or disable the protection it wants to bypass.  The real problem is just that if it can do this, so can anything at all, which completely defeats the purpose of even having the protection mechanisms in the first place.  They're just extra code and complexity for nothing if they do nothing.  But I absolutely agree with the original ideals that such basic protections should be in place.

As for management, there's always just shutting down and taking out the memory card and modifying files that way.  Do people really need to swap modules around a lot?  I would think that generally custom modules are a once in a blue moon kind of thing if even that often (last time I personally used one was for Incognito so I could use homebrew online back when Atmosphere didn't have an INI setting to disable its protection and now even that has an INI setting so no module needed anymore.)  I imagine a lot can be done via bootloader configurations in Hekate or similar anyway.

I assume something like a sudo type system (eg just ask the user for a PIN) would be too much to implement, right?  Though that sure would be nice too.  That would prevent apps from just doing whatever without asking first at least.

skalfate, 2020-04-29:
I agree with @SciresM Scires on this. 

The OP's fear of malicious activity comes from installing a source he can't verify. A source that repeatedly misaligns itself with atmosphere. Then it is a users decision/responsibility to verify what homebrew they want to install and from a trusted source.

No matter the measures of protection, if there is a feature a user is chasing, he will accept any method offered to defeat the protection to achieve the feature they want. Such as forks and solutions that brought you to open this request. So in the end, what is best is to provide the essential protection of catastrophic failure, which is atmosphere creating backup of keys and prodinfo.

In any case the solution of pin protection is creating another layer to the very system we currently have. Imagine had we had pin protection, you would of installed the very same app, entered you pin, and realized if this app can do it so can any other.

shchmue, 2020-04-29:
how do you suggest blocking an app, say, faking a crash screen then rebooting to a bundled payload that does malicious stuff

also putting stress on the SD reader-board connection is to be avoided as much as possible

SciresM, 2020-04-29:
I am going to go with "It kind of seems to me like the best solution is to prevent anything catastrophic from occurring (this is the bis protection policy in general), and then ask people to only run homebrew from sources they trust.".

closed 2020-04-29 by SciresM

Nazosan, 2020-04-29:
Ok, but this is like having a lock on a door -- that's on both sides.  Just extra complexity that doesn't really achieve anything since the person can turn the lock from the other side anyway.  You may as well remove the built-in protections since anything can bypass them in a single move now.

SciresM, 2020-04-29:
"tinfoil" I have already said warez tools are explicitly not supported.

"You may as well remove the built-in protections since anything can bypass them in a single move now.".

Okay. Please explain the mechanism by which a homebrew damages the console unrecoverably, under the current protection scheme? I claim this is mostly a non-issue.

Also, there is a fundamental conflict between a homebrew ecosystem's desire to be permissive and let users do what they want with their devices and to prevent users from shooting themselves in the foot. Some kind of compromise is inevitable. There is no perfect solution.

friedkeenan, 2020-04-29:
I *mean*, a homebrew could always reboot to a malicious payload and cause loads of fuckery from there, but that's probably not worth being in the threat model or whatever

SciresM, 2020-04-30:
That's deeply not in the threat model that I care about, yeah.