Atmosphere-NX/Atmosphere #938: Error when exiting Animal Crossing.

MySTiXX666, 2020-04-30:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

We as a family were playing Animal Crossing. Every once in a while when exiting/closing Animal Crossing we get an error. The error report shows Program ID: 010000000000000d and Proces name: dmnt. That should be the cheat module. I am using Edizon and it's overlay and a cheat for Animal Crossing, but even when not using the cheat the error occurs. No always but the last couple of days a few times. It happened when a family member was added to the Island. But that could be a coincidence.

### How can the issue be reproduced?

It not always happens, so it occurs without any lead. Even when not using the overlay or cheat for that matter it happens.

### Crash Report

https://gist.github.com/MySTiXX666/1a95ee15847725e7817483dc749a9a84.js

### System Firmware Version

10.0.2

### Environment?

Using Atmosphere 0.12.0 and directly injected.

### Additional context?

Not that I know off any.

exentio, 2020-05-01:
I'm having frequent crashes while playing ACNH and opening the HBL as applet. No sysmodules, no cheats

evertonstz, 2020-05-02:
> I'm having frequent crashes while playing ACNH and opening the HBL as applet. No sysmodules, no cheats

Don't know if it's related with OP"s, but I've just filled a report at #941 that seems pretty close to what you're having.

SciresM, 2020-05-02:
@evertonstz Your stuff isn't related.

This guy's crash is due to some changes I made to the cheat engine.

Previously to 0.12.0, we had a guarantee about the states the debug event continuation thread could be in on process exit, but some of these invariants broke when I added fake pause/resume to the cheat engine; this issue (0x41001) occurs from the cheat engine code asserting that an operation will succeed on the cheat process but the operation fails because the process has been closed (as it being open is no longer an invariant at that point in time).

This issue just needs me to spend like an hour or two figuring out which specific call has the invariant broken and adding some handling to that case.

Patrick5897, 2020-05-16:
So with the previous atmosphere version everything works fine? I want to update to 10.0.2 but I use cheats for ACNH.

commit 9baf096a (2020-05-16): dmnt: make debug event result handling more robust (closes #938)
closed 2020-05-16 by SciresM