Atmosphere-NX/Atmosphere #939: Add a option for animated boot screens

Lava3063, 2020-04-30:
I saw two good animated boot screens and was looking to find a way to add them, only to come to the disappointment that there was none. Please implement them if possible, or add an option for us to add them ourselves. Here are the two.

[https://imgur.com/a/iIEpb](https://imgur.com/a/iIEpb)

[https://www.youtube.com/watch?v=5eQtJNm2mWo](https://www.youtube.com/watch?v=5eQtJNm2mWo)

Thank you!

SciresM, 2020-04-30:
Please use the issue templates in the future.

Animated boot screens will not be implemented, as most of the time booting is spent preparing for boot. The Atmosphere splash screen shows only really while work is being done in the background. Spending CPU time rendering the animated screen would be a lot of work for unnecessary boot delay.

If you implement an optional animated boot screen feature that doesn't cause noticeable boot delay, feel free to make a pull request.

closed 2020-04-30 by SciresM

bakugo, 2020-04-30:
>The Atmosphere splash screen shows only really while work is being done in the background

Why is it set to always show for at least 3 seconds then?

SciresM, 2020-05-01:
@bakugo Because it takes ~2-2.5 seconds on my switch for all the loading logic to be done, and I think that it is a better choice to have a consistent time whenever possible. 

Three seconds is enough for all the background processing to complete, and the delay means that we always show the screen for the same amount of time.