Atmosphere-NX/Atmosphere #941: Almost daily crashes that seem to be related with hbmenu, started after 0.12.0 update

evertonstz, 2020-05-02:
## Bug Report

### What's the issue you encountered?

Atmosphere crashes at random, but it seems to happen more when opening or exiting a homebrew at hbmenu, but actually happened once or twice while pressing home from a game.
I have no changes on atmosphere. This all started after I update to the lateste version 0.12, also, I've checked and hbmenu.nro is not corrupted as the checksum matches the one in atmos zip file and I did a new install of atmos just be sure, didn't solve it.

### How can the issue be reproduced?

I'm not sure, the problem seems to happen at random, but I've noticed it happens more when exiting or opening any homebrew at homebrew menu, but also happens on exiting a game, like BOTW.

### Crash Report

This is the latest crash report: https://pastebin.com/D002Dr2W
This one happened with a tesla overlay, btw: https://pastebin.com/f1pA9bgW
And this one seem to have a different error code: https://pastebin.com/5gpJ8BkF
I've others but they're basically the same, they're all from the same day, if needed just ping me and I'll upload all the error files.

### System Firmware Version

10.0.1

### Environment?

- bootloader: fusee-primary
- official atmosphere release 0.12.0
- sysmodules: tesla and sys-screenshot
- Homebrew software installed: goldleaf, retroarch and edizon

SciresM, 2020-05-02:
@evertonstz "Instruction abort in program `0104444444441001`"

You are using an unofficial/not supported version of hbl. Support can't be provided for that.

Data abort in ovloader <-- sounds like not an atmosphere bug also

assertion failure in bsdsockets <-- only one of the three that could be real but I'm inclined to suggest some homebrew you have in the background is causing it...

evertonstz, 2020-05-02:
> You are using an unofficial/not supported version of hbl. Support can't be provided for that.

Can you explain more on this? What exactly did I do to use this unofficial version of hbl? My atmosphere install came straight from the releases in this repository, but I think my last update came from atmosphere updater. Could it be Tesla? 

ovloader is a test module for tesla overlay.

> assertion failure in bsdsockets <-- only one of the three that could be real but I'm inclined to suggest some homebrew you have in the background is causing it...

Sure, but like I said, I made a clean install yesterday, and the bug still happens, I have no other homebrew installed currently.

SciresM, 2020-05-02:
"0104444444441001" <-- this is a custom program ID. This means that you installed an unofficial program/nsp.

Warez/content installation is generally not supported at all.

It sounds like your install really isn't actually clean.

Does issue occur in clean atmosphere without additional sysmodules (no tesla/sys-screenshot, etc)?

evertonstz, 2020-05-02:
> "0104444444441001" <-- this is a custom program ID. This means that you installed an unofficial program/nsp.
>
> Warez/content installation is generally not supported at all.
>
> It sounds like your [...]

Could be from this: https://gbatemp.net/threads/homebrew-menu-loader-legal-nsp.518433/ ?
It's the only thing I've actually found matching this ID and I actually had this NSP installed, but I wasn't launching HBmenu from that, I was launching it from applet mode.

But like I said: I currently have a clean install, no homebrew, no NSP installed, nothing, and I'm still getting crashes related with "bsdsocket".

Anyways, I'm gonna do another clean install and check things out.

I'm closing the issue and will reopen with new logs in case things don't get better.

closed 2020-05-02 by evertonstz