Atmosphere-NX/Atmosphere #943: fatal error occured when running atmosphere (titleid 0100000000000005)

Servietska, 2020-05-05:
NOTE : this is the same issue you solved for AaronGeorge ("Atmosphere Fatal Error on Boot #752"); but the solution you gave him doesn't work (use Archive bit fixer in Hekate)

### What's the issue you encountered?

A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Did you make any changes related to Atmosphère itself? 
No

### How can the issue be reproduced?

Boot to RCM
Launch Hekate
Boot Atmosphere

### Crash Report
https://www.mediafire.com/file/o6x6kbqb1nufae2/report_0000000005d8646a.bin

### System Firmware Version
10.0.2 (but the stock system still ask me for an update)

### Environment?

Bootloader Hekate (5.2)
Atmosphere (0.12.0) - Official
Using Edizon occasionnaly

### Additional context?

I didn't used my switch for 2 months, and there was at least 2 fimrwares updates since.
So I proceeded updating to 10.0.2 with ChoixDujour. Everything seems to be OK.
After reboot, I got a message that Atmosphere was outdated.
So I proceeded updating both Hekate (from 5.0.2 to 5.2) and Atmosphere to 0.12.0)
And since I got the Fatal Error

I try the archive bit fixer ==> didn't work
I try deleting  all folder on my SD and copy again the folders from SDSETUP (additionnal question : can I format or or the Switch is linked to THIS SD card ?)

Thanks a lot for your lights

SciresM, 2020-05-05:
You have an SD card formatted as exFAT but installed non-exFAT firm.

Format your SD card as FAT32.

closed 2020-05-05 by SciresM

Servietska, 2020-05-05:
Hi sir, thx a lot. WOrking again. \o/
For anyone having the same issue, I formatted the 64 go card with Fat32 Format tool. (every other tool I try before your answer didn't work, after your answer I kept seeking a working tool and here I am :))

hoskinsw, 2020-05-07:
I am having the exact issue, but my SD card is in FAT32 format. I have gotten this to work before, but after formatting the sd card and trying to use 0.12.0 I have this problem come up.

One weird thing is that the SD card gets corrupted after I power the switch down and take it out.

Another thing that is strange is that I get this same error in exFAT and FAT32, but in FAT32 it takes a few minutes for it to happen. While booting the screen stays black for a while then the error pops up, unlike in exFAT where it comes up almost immediately.

Didn't want to open another issue since it is basically the same one, thanks!

hstaoqian, 2020-07-13:
> Hi sir, thx a lot. WOrking again. \o/
> For anyone having the same issue, I formatted the 64 go card with Fat32 Format tool. (every other tool I try before your answer didn't work, after your answer I kept seeking a working tool and here I am :))

Hi, I encounter the same problem. Did you slove the problem? Thanks for your reply.