Atmosphere-NX/Atmosphere #944: Atmosphere error

lovricm1805, 2020-05-06:
Hello guys, Im fairly new to switch hack scene, so in front i am sorry.

Okay i installed cfw following that site with main guide ( one for dummies )

im injectin hekate

few days it worked normally, now for first time i played it docked, and suddenly some error has been shown, and i could see through dock some yellow error messages. I injected payload again, tried edizon to put in cheats in Pokemon sword ( i already installed game passes i dont know if that causes the problem )

edizon failed, i will send in attachment the picture of error.

now when i enter pokemon, go to menu, it automatically forces me out of world map, items etc. And crashes with same error. Btw pokemon is played with cartridge

next i went to play some nsp game, mortal kombat.. and it crashes after few minutes, i pressed volume button to send me automatically to payload.. but then system vibrated weirdly.

please help me

Ps. I cant post photo, please check it on this link where i posted same stuff https://gbatemp.net/threads/switch-atmosphere-error.564480/

inspier, 2020-05-06:
If you are using a recent version of Hekate did you remember to update the bootloader on your SD card when you updated (just a guess based on what you said about yellow text and the rumble). Also, warez/content installation is not condoned nor supported.

lovricm1805, 2020-05-06:
I never updated anything.. everything was fine until today

SciresM, 2020-05-16:
Please re-open this using the issue templates.

closed 2020-05-16 by SciresM