Atmosphere-NX/Atmosphere #945: Custom NSPs broken

IISAKNIFE, 2020-05-07:
## Bug Report

### What's the issue you encountered?

Custom NSPs, such as those made with nro2nsp don't work on the latest atmosphere</br>
### How can the issue be reproduced?

Install an NSP made by nro2nsp, and try to launch it on Atmosphere 12</br>

### System Firmware Version

10.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official 0.12.0<br>
### Additional context?
Sigpatches don't affect this issue. I was able to launch the backups that I made from my gamecards just fine. I am only unable to launch custom NSP applications

fincs, 2020-05-07:
The sole supported way of running homebrew is through NRO loaded by nx-hbloader. Everything else is completely unsupported, extremely harmful, and requires piracy enabling patches.

closed 2020-05-07 by fincs

IISAKNIFE, 2020-05-07:
> The only supported way of running homebrew is through NRO loaded by nx-hbloader. Everything else is completely unsupported, extremely harmful, and requires piracy enabling patches.

How is me doing what I want with my switch in ANY WAY harmful?