Atmosphere-NX/Atmosphere #946: Crash at boot on titleid 0100000000000005

AaronClover, 2020-05-08:
## Bug Report

a fatal error occured when running atmosphere
title id: 0100000000000005
error desc: std::abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_00000000012b2cb8.bin

### What's the issue you encountered?

Updated sysnand to 10.0.2 and initialized with ChoiDujorNX successfully, I did the same to emunand and ran into the error below when trying to load Atmosphere on sysnand or emunand.

### How can the issue be reproduced?

See above.

### Crash Report

https://www.mediafire.com/file/9j7eduge3lvj1im/report_00000000012b2cb8.bin/file

### System Firmware Version

10.0.2

### Environment?

Using Fusee payload
Atmosphere 0.12.0
Hekate 5.2.1

### Additional context?

Booting stock sysnand normally results in black screen, but can boon stock sysnand from hakate.

Only have a BOOT 0 & BOOT 1 backup from 9.1.0. Restoring did not change the issue.

I have attempted booting with exfat and fat32 using gui fat32format and get the same error.

I have attempted to wipe the sd and use a clean install of hekate + atmos and get the same error.

I attempted to reinstall the firmware with ChoiDujour.exe but i'm getting "Failed to match key" for every key.

Steve0the0cube0man, 2020-05-08:
Bruh they haven't added support for 10.0.2. Of course you gonna get crashed. also, booting sysnand black screen means you have auto rcm on.

ThatNerdyPikachu, 2020-05-08:
> Bruh they haven't added support for 10.0.2. Of course you gonna get crashed.

10.0.1 and 10.0.2 only changed things that do not effect the compatibility of Atmosphere with those firmwares.

Steve0the0cube0man, 2020-05-08:
ahh ok thanks

Steve0the0cube0man, 2020-05-08:
weird,i have the same problem

SciresM, 2020-05-08:
This is a "nonsense" crash that makes no sense -- crash indicates svcQueryIoMapping is returning an error code that it never should.

Try using a clean atmosphere folder?

The only way I can imagine this happening is if you "installed" 10.0.0 but 10.0.0 package1/package2 isn't written to emmc. This could cause this behavior but it's corrupt nand, so there's not much we can do...

EliseZeroTwo, 2020-05-08:
I encountered a similar issue with someone who had upgraded through Choi to 10.0.1 from some firmware on a different key generation (8.0 if I'm not mistaken) but in their case Choi didn't upgrade their PKG2 so hekate was failing to decrypt it on boot.

System was upgraded though. We fixed it by reflashing bcpkg2-(1/2/3/4) through nxfw-tool.

SciresM, 2020-05-08:
Yeah, this would cause this. 10.x changed the kernel ABI for QueryIoMapping. Gonna close issue since it's a "user has corrupted fw" problem.

closed 2020-05-08 by SciresM

AaronClover, 2020-05-09:
Somehow the boot became corrupt. Was able to restore boot0 and boot1 from emunand and all is working. I think I may have selected exfat for sysnand and then fat32 for emunand when using ChoiDujourNX.

EliseZeroTwo, 2020-05-09:
> Was able to restore boot0 and boot1 from emunand and all is working.

Glad you fixed it! 

> I think I may have selected exfat for sysnand and then fat32 for emunand when using ChoiDujourNX.

This doesn't matter as long as the SD card you use when booted into emunand is fat32, and even then it wouldn't cause this.

jerry3547, 2020-05-21:
I had tried to delete all files in atmosphere folder, but still got the error