Atmosphere-NX/Atmosphere #948: Panic error when launching hbmenu via album.

bcoliver, 2020-05-09:
## Bug Report

### What's the issue you encountered?
I receive a panic error when attempting to launch hbmenu through the album

### How can the issue be reproduced?
- Launch emuMMC through Hekate
- Open Album

### Crash Report
[report_0a4d26bf.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4602401/report_0a4d26bf.zip)

### System Firmware Version

10.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build:  Official 5.2.1
 
- Do you have additional kips or sysmodules you're loading: Kosmos, if that counts (sorry - new to this)
- Homebrew software installed: SkyNX

### Additional context?

- Additional info about your environment: Today I installed Atmosphere through the homebrew guide, and everything worked fine to begin with. However, I noticed that when I would try and launch a game through my emuMMC, I would get an error about necessary game data being missing. I copied the Nintendo folder on the root of my SD card and merged it with the one contained within the emuMMC folder. This did allow me to play my games on the emuMMC but for whatever reason, opening hbmenu returns a panic error. I then reverted the changes to the Nintendo folder in an effort to fix it, but now games will not launch and the panic error persists when I open hbmenu.

I apologize for the formatting errors in my original Issue, logging into my account for whatever reason allowed me to bypass the templates.

SciresM, 2020-05-09:
No worries about formatting stuff, it happens.

The error report is in title id 0x690000000000000D. This is not an atmosphere title id, googling says it's sys-con.

Remove/disable sys-con, as it's what's crashing. Should solve the problem.

To do this, remove the /atmosphere/contents/690000000000000D folder. Confirm this fixes problem, and I'll close issue.

bcoliver, 2020-05-09:
Apologies for the late response, I was just out for dinner.

After deleting sys-con, everything seems to be working normally again. Thank you so much for your help!

SciresM, 2020-05-09:
np, glad it's resolved.

closed 2020-05-09 by SciresM