Atmosphere-NX/Atmosphere #950: Atmosphere wont load all the way; black screen

Coolaaron88, 2020-05-10:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

After upgrading from CFW 9.2 to CFW 10.0.2 I started having this issue. Atmosphere wont start up all the way. After Atmopshere logo etc i have a black screen for couple secs and an error.

### How can the issue be reproduced?

Downloaded latest 0.12.0 update for Atmosphere, including fusee-primary.bin
Copied every file to SD card, fusee-primary.bin to atmosphere folder.
When asked if I want to overwrite files, I did that.
Plugged in SD card,
Turning on console with RCM loader and jig installed

### Crash Report

A fatal error occurred running Atmosphere.
Title ID: 010000000000002b
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/

[report_000000000778d152.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4604991/report_000000000778d152.zip)

### System Firmware Version

10.0.2

### Environment?

Atmosphere launched by fusee.bin using TegraRCM

### Additional context?

- Additional info about your environment:
- SD card is formatted as ExFat and not Fat32

SciresM, 2020-05-10:
You have archive bits set on atmosphere/atmosphere subfolders, this is causing path not found to be returned when ensuring the erpt_reports directory. Unset the archive bits.

closed 2020-05-10 by SciresM

Coolaaron88, 2020-05-10:
Thank you for that