Atmosphere-NX/Atmosphere #951: "Unable to start software" on specific titles.

Monkeor, 2020-05-10:
I was on firmware 9.2.0 using atmosphere for a while, but trying to install the SM64 nsp that's been recently floating about using goldleaf gave me an error. I updated my console's firmware to 10.0.2 using ChoixDuJour, but forgot to update my Atmosphere or sigpatches. The console booted into atmosphere just fine, though trying to install the nsp still gave me the error, and trying to play any of my installed games also gave an "Unable to start software" message. I then remembered to update atmosphere and put in sigpatches for 10.0.0, which let me install the nsp as well as play my games again.

Unfortunately, now when I try and run the SM64 nsp, I get an "Unable to start software" error. I installed the AM2R nsp, but trying to run it gave the same error. For testing, I uninstalled one of my games which does work, then reinstalled it to see if it would give me the same error. It didn't, and it worked just fine. So far, as much as I can tell, only SM64 and AM2R cannot be played, and I can't find out what's wrong. I could really use a bit of help, and I hope this isn't the incorrect place to ask about this.

birbb, 2020-05-10:
Not supported nor does it relate to Atmosphere either.

Monkeor, 2020-05-10:
So there's no chance I could have screwed something up when it comes to my installation or configuration of Atmosphere? Another Atmosphere user I spoke to hasn't encountered any issues trying the same thing, so my assumption was that I'd somehow broken something.

Monkeor, 2020-05-10:
Yep, it's a matter of CFW, not something having broken. Switching CFW solved the issue.

closed 2020-05-10 by Monkeor