Atmosphere-NX/Atmosphere #953: Panic occured while running atmosphere

phranci, 2020-05-12:
### What's the issue you encountered?

Panic occured while running atmosphere

TitleID: 010041544D530000
Error Desc: std:abort() (0xFFE)

### How can the issue be reproduced?

While having a game running in the background, launch hbmenu
closing the game before launchin the hbmenu fixes the problem

### Crash Report

https://github.com/phranci/report/blob/master/report_031b4f99b.bin

### System Firmware Version

9.0.0</br>

### Environment?

What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
Official release or unofficial build: Official release version - Atmosphère 0.12.0
Do you have additional kips or sysmodules you're loading: N/A
Homebrew software installed: N/A

### Additional context?

Atmosphère 0.10.5 working fine, updated to 0.12.0 and the problem started to appear.
already tried: 

fresh install on new SD card fat32 formatted
updated FW to 10.0.1 exFAT via ChoiDujourNX
fresh Atmosphere install on the same sd card
hekate archive bit fix

edgar1016, 2020-05-14:
Having the same problem expect mine only happens after installing the CFW waiting a few hours then rebooting.  While booting up the error occurs while booting. It also corrupts my SD card, I then have to format the card reinstall atmosphere and hekate recreate the emuMMC only to finally boot back into the CFW rebooting and doing it all over again. I thought I had a fake SD card so I ran H2testw it all came back clean. My OFW is on 8.1.0 the first thing I do after creating the emuMMC is upgrade to 10.0.2. This is all on windows 10 as well.

jerry3547, 2020-05-21:
I'm also got the same error when try to launch the payload
[report_05b8caa4.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4662588/report_05b8caa4.bin.zip)

phranci, 2020-05-21:
> I'm also got the same error when try to launch the payload
> [report_05b8caa4.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4662588/report_05b8caa4.bin.zip)
> 
> after reading other post, I found the way to fix this error
> 
> 1. check the SD card format is exFat or Fat
> 2. re-format the SD card into Fat

it's a different issue. reformatting the sd card is not working

PrimetimeFluff, 2020-05-27:
I have the same issue @phranci

phranci, 2020-05-28:
> I have the same issue @phranci

Unfortunately I wasn't able to find any working solution. The switch sits on a shelf atm, I will wait the next official release hoping it will fix the problem I guess.

jerry3547, 2020-05-28:
> > I have the same issue @phranci
> 
> Unfortunately I wasn't able to find any working solution. The switch sits on a shelf atm, I will wait the next official release hoping it will fix the problem I guess.

did you try to format the SD card to FAT32?

SciresM, 2020-05-28:
Not an atmosphere issue.

Sysmodules are exhausting the FS session limit (it appears most custom sysmodules outside atmosphere are not configuring themselves to only use one FS session instead of three, and are counting triple against the limit).

This is a fixed, unpatchable system limitation -- the developers of the sysmodules you use will need to release better behaved builds which correctly respect the limitations.

I'm reaching out to some sysmodule developers (e.g. https://github.com/exelix11/SysDVR/issues/58) to get them to do this where I can.

closed 2020-05-28 by SciresM