Atmosphere-NX/Atmosphere #954: System Crash when Sleeping & Powering Off, other odd behavior

HeroCC, 2020-05-14:
## Bug Report

### What's the issue you encountered?

I updated from an old SDSetup bundle (I forget exactly which atmosphere it contained, but it was a while ago, I *think* near version 0.8.9; I have an SD card backup if it is important) to the most recent one as of yesterday (containing version 0.12.0, tried with and without default sysmodules). No changes aside from the default recommended settings, if you'd like me to make any to debug I'd be happy to. Now, whenever I put the system to sleep or attempt to power-off, a crash with error code `0162-0002 (0x4a2)` occurs. 

### How can the issue be reproduced?

* Using Hetake, boot to Nyx via RCM injection by fusee-launcher
* Select Launch -> CFW (Sysnand)
* System boots as usual...
* Attempt to enter sleep mode via any method (wait, select sleep mode) OR hold power button and select power off
* Screen goes black, then powers LCD but remains black, both very quickly, then goes to the [Atmosphere crash screen](https://user-images.githubusercontent.com/2124489/81888570-9dd34a80-956f-11ea-9269-9c948367c509.png)

### Crash Report

I cleared all of the crash / fatal reports, and after one occurrence these files were there:
* `fatal_reports/01589424121_0100000000000021.log`: https://pastebin.com/ZNfEScTB
* `crash_reports/01589424121_0100000000000021.log`: https://pastebin.com/pcYwkvhv
* `crash_reports/01589424122_0100000000000021.log`: https://pastebin.com/ahDe93eA
I find it odd that the crash reports had non-ascii chars, is that related?

### System Firmware Version

5.1.0 (via ChoiDujourNX)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate v5.2.1
- Official release or unofficial build: 
  - Kosmos / SDSetup downloaded May 13th
- Do you have additional kips or sysmodules you're loading: kip`default_nogc`, and default SDSetup sysmodules, but the problem occurs with them removed as well
- Homebrew software installed: Nothing special, but on an aside I can't access HBM either -- it stops at a black screen (but still powered on)

### Additional context?

When launching the Homebrew menu from the Album button, Atmosphere crashes in a different way, saying "[Panic occurred while running Atmosphere](https://user-images.githubusercontent.com/2124489/81888651-cf4c1600-956f-11ea-86ee-09091a8a7d4f.png)". I can attach that report if you'd like it too. Again, not sure if this and the odd crash reports are related, but my priority ATM is being able to put my system to sleep safely.

Thank you so much for your great project, let me know if you need any more information!

SciresM, 2020-05-14:
Known issue on exactly 5.1.0, fixed via 8e75a4169d1355bcaf4e36db70acd5a332578497 and 0e289461bb6b9a0baf3e1ff680557e531326e38f.

Wait for the next atmosphere release, or if it's urgent downgrade to 5.0.2 or upgrade to 6.0.0.

closed 2020-05-14 by SciresM

HeroCC, 2020-05-14:
:heart: you're a god among men, tysm!