Atmosphere-NX/Atmosphere #955: 010000000000002B

mplu, 2020-05-15:
## Bug Report
A fatal error occurred when running Atmosphere.</br>
Title ID: 010000000000002B</br>
Error Desc: std::abort() called (0xffe)</br>
Report saved to /atmosphere/fatal_errors/report_3af9b353d.bin</br>

[ * ]</br>
Run kosmos using applet mode</br>

### What's the issue you encountered?

When I press the power button to go in sleep mode, or when the console go to sleep mode by itself</br>

### How can the issue be reproduced?

[ * ]</br>
Presse the power button to put in sleep mode

</br>

### Crash Report
[report_3af9b353d.zip]</br>

### System Firmware Version

5.1.0 | AMS 0.12.0 | E</br>

### Environment?

- What bootloader :  hekate : 
     Kosmos' distribution of Atmosphère : 16.1

### Additional context?

- Happend since I make a partion for emummc, but also since update from ATM/Kosmos.

mplu, 2020-05-15:
[report_3af9b353d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4634265/report_3af9b353d.zip)

SciresM, 2020-05-15:
Duplicate of a few other issues. Already fixed in master.

Wait for the next atmosphere release, or downgrade to 5.0.2 or upgrade to 6.0.0.

closed 2020-05-15 by SciresM