Atmosphere-NX/Atmosphere #957: Remove screenshot limits

GRAnimated, 2020-05-16:
## Feature Request

### What feature are you suggesting?
#### Overview:
I would like a way to take as many screenshots and videos as I want without the limits. Nintendo's system memory limit is 100 for videos and 1000 for screenshots, and the SD card limit is 10000 screenshots iirc. From the 3 years I have had my switch, I've filled up my screenshots and now I'm at the limit.

#### Smaller Details: 
N/A

#### Nature of Request:  
- Addition
  - Remove Nintendo's screenshot and video limits.
- Change
  N/A
- Removal
  N/A

### What component do you feel this would best fit within?
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
It's more of a convenience thing so you don't have to clear out any screenshots.

SciresM, 2020-05-16:
Implement it yourself and PR. This is otherwise a waste of time because you can just move your screenshots to your PC.

closed 2020-05-16 by SciresM

HookedBehemoth, 2020-05-28:
What fw version are you on?

HookedBehemoth, 2020-05-28:
If on 10.0.0 could you give this one a try?
It overwrites all limits with 65535. Patches [this function.](https://github.com/HookedBehemoth/capsrv-reimpl/blob/overhaul/capsrv/source/capsrv_album_settings.cpp#L151)

https://0x0.st/ifZZ.zip

HookedBehemoth, 2020-06-17:
Remembered that while this does remove the limit, the album applet will still only request/show 11000 files from sd and 1100 from nand.

ehx-v1, 2020-08-16:
>  This is otherwise a waste of time because you can just move your screenshots to your PC.

The fact the screenshot/video limit is reached doesn't show until one wants to make one more capture. IMO the file limits themselves are less of a problem than not warning early.

Grass0001, 2020-08-16:
who has 10k screenshots that they are never going to move over 3 years?

HookedBehemoth, 2020-08-16:
Reminder that this is an issue tracker not a discussion forum.