Atmosphere-NX/Atmosphere #959: Fatal auto reboot doesnt work properly

fennectech, 2020-05-17:
## Bug Report

### What's the issue you encountered?

The auto reboot doesnt work properly  Ive found it reboots instantly on a value of 0x5000 and any other values ive tried show fatal screen indefinitly

### How can the issue be reproduced?
Set fatal_auto_reboot_interval = u64!0x5000 and trigger a fatal  The console will reboot instantly
Use any other value  it will never reboot 
### System Firmware Version

10.0.2</br>
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate
- Official release or unofficial build: 
  - [ Official release version 0.12.0 ]
- Do you have additional kips or sysmodules you're loading:No
- Homebrew software installed: none

### Additional context?

Plane jane atmosphere (booted with hekate) on a newly formatted sd card  No custom sysmodules not added by atmo

Slluxx, 2020-05-22:
also a value of 0x3E8 doesnt work.

SciresM, 2020-08-04:
Closed by 501280b6e5043ce6a2d1ab0c2f8918ca5786b05c.

closed 2020-08-04 by SciresM