Atmosphere-NX/Atmosphere #964: Build documentation issue or master is not building

PabloZaiden, 2020-05-25:
## Bug Report

### What's the issue you encountered?
When trying to build current `master` branch (at commit `f215da`) it starts building for a couple of minutes, until I get the following error:

```
Atmosphere/libraries/libstratosphere/source/os/impl/os_internal_critical_section_impl.os.horizon.cpp:51:22: error: '::mutexIsLockedByCurrentThread' has not been declared; did you mean 'IsLockedByCurrentThread'?
   51 |             return ::mutexIsLockedByCurrentThread(std::addressof(this->thread_handle));
```
### How can the issue be reproduced?
- Open a development environment. In this case, I'm using an Ubuntu-based container, but I've reproduced the same issue on a Macbook with OS X Catalina

Dockerfile:
```Dockerfile
FROM ubuntu
ARG DEBIAN_FRONTEND=noninteractive
RUN apt update
RUN apt install -y wget git build-essential libxml2 libxml2-dev libxml2-utils
WORKDIR /
RUN wget https://github.com/devkitPro/pacman/releases/download/devkitpro-pacman-1.0.1/devkitpro-pacman.deb
RUN dpkg -i devkitpro-pacman.deb
RUN dkp-pacman -S --noconfirm switch-dev switch-libjpeg-turbo devkitARM devkitarm-rules
RUN rm devkitpro-pacman.deb
ENV DEVKITPRO=/opt/devkitpro
ENV DEVKITARM=/opt/devkitpro/devkitARM 
```
- Clone Atmosphere
- Run `make`

### Additional context?

As said, the same error appeared building on a Macbook and inside a Docker container with the requirements from [building.md](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/building.md)

masagrator, 2020-05-25:
Atmosphere repo between releases is often broken, so it's not adviced to work on fresh commits. My advice is to wait for new release as 0.12.0 is not compatible with new gcc.

SciresM, 2020-05-25:
Yeah. The only thing I can really guarantee is that atmosphere-master builds with libnx-master and latest devkitA64.

PabloZaiden, 2020-05-25:
Will the next release be compatible?

SciresM, 2020-05-25:
The specific issue you mention requires libnx-master, you can clone and make install libnx to build it.

The next release will still require libnx-master at the time -- atmosphere and libnx kind of evolve side by side.

Current atmosphere master is buildable if you make install master libnx.

ZachyCatGames, 2020-05-25:
You also need libnx master
(Which can be gotten by cloning https://github.com/switchbrew/libnx, running `make`, then `sudo -E make install`)

PabloZaiden, 2020-05-25:
Thanks. I'll build libnx master first then.

If possible, consider creating docker image with tags aligned with the releases having a snapshot of the dependencies to build each version. Especially if the code depends on the master branch of a different repo.

Thanks again for the pointers!

SciresM, 2020-05-25:
I'm kind of opposed to docker images or anything before atmosphere 1.0.0 (which is "probably" this year, when mesosphere is working). Glad you've got it solved though.

closed 2020-05-25 by SciresM